erpalma/throttled #6: Script crashes on Ubuntu 18.04

marcinsulikowski, 2018-05-13:
I tried using the `lenovo_fix` service on T480s running Ubuntu 18.04 (Linux kernel 4.15) but it fails when starting:

```
user@t480s:~$ systemctl status lenovo_fix.service
● lenovo_fix.service - Fix Lenovo T480/T480s/X1C6 throttling on Linux
   Loaded: loaded (/etc/systemd/system/lenovo_fix.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-05-13 14:35:56 CEST; 7min ago
  Process: 1399 ExecStart=/opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py (code=exited, status=1/FAILURE)
 Main PID: 1399 (code=exited, status=1/FAILURE)

May 13 14:35:56 t480s python3[1399]:     main()
May 13 14:35:56 t480s python3[1399]:   File "/opt/lenovo_fix/lenovo_fix.py", line 141, in main
May 13 14:35:56 t480s python3[1399]:     undervolt(config)
May 13 14:35:56 t480s python3[1399]:   File "/opt/lenovo_fix/lenovo_fix.py", line 61, in undervolt
May 13 14:35:56 t480s python3[1399]:     writemsr(0x150, calc_undervolt_msr(plane, config.getfloat('UNDERVOLT', plane)))
May 13 14:35:56 t480s python3[1399]:   File "/opt/lenovo_fix/lenovo_fix.py", line 37, in writemsr
May 13 14:35:56 t480s python3[1399]:     os.write(f, struct.pack('Q', val))
May 13 14:35:56 t480s python3[1399]: PermissionError: [Errno 1] Operation not permitted
May 13 14:35:56 t480s systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=1/FAILURE
May 13 14:35:56 t480s systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.
```
The full error messages is as follows:
```
root@t480s:~# source /opt/lenovo_fix/venv/bin/activate
(venv) root@t480s:~# /opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py
Traceback (most recent call last):
  File "/opt/lenovo_fix/lenovo_fix.py", line 169, in <module>
    main()
  File "/opt/lenovo_fix/lenovo_fix.py", line 141, in main
    undervolt(config)
  File "/opt/lenovo_fix/lenovo_fix.py", line 61, in undervolt
    writemsr(0x150, calc_undervolt_msr(plane, config.getfloat('UNDERVOLT', plane)))
  File "/opt/lenovo_fix/lenovo_fix.py", line 37, in writemsr
    os.write(f, struct.pack('Q', val))
PermissionError: [Errno 1] Operation not permitted
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/opt/lenovo_fix/venv/lib/python3.6/site-packages/periphery/mmio.py", line 61, in _open
    fd = os.open("/dev/mem", os.O_RDWR | os.O_SYNC)
PermissionError: [Errno 1] Operation not permitted: '/dev/mem'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/lenovo_fix/lenovo_fix.py", line 113, in power_thread
    mchbar_mmio = MMIO(0xfed159a0, 8)
  File "/opt/lenovo_fix/venv/lib/python3.6/site-packages/periphery/mmio.py", line 36, in __init__
    self._open(physaddr, size)
  File "/opt/lenovo_fix/venv/lib/python3.6/site-packages/periphery/mmio.py", line 63, in _open
    raise MMIOError(e.errno, "Opening /dev/mem: " + e.strerror)
periphery.mmio.MMIOError: [Errno 1] Opening /dev/mem: Operation not permitted
```

marcinsulikowski, 2018-05-13:
It seems that the issue is caused by the Linux kernel lockdown (https://lwn.net/Articles/706637/) which is typically triggered by the EFI Secure Boot feature. I disabled Secure Boot in BIOS settings and everything started working as expected. Nothing is actually broken in the `lenovo_fix` script but the `Readme.md` file could say that disabling Secure Boot may be needed (or even the script itself could suggest this when it encounters the issue).

erpalma, 2018-05-13:
You are right, right now I have secure boot disabled so I didn't tested my script with it. I'll point out this "limitation" in the readme. Thank you for pointing out!

erpalma, 2018-05-14:
Readme updated. Thank you!

closed 2018-05-14 by erpalma

daenney, 2019-01-15:
I have secure boot enabled (with my own signing keys) and everything seems to work just fine. Kernel in question is 4.20.1 so this might not be necessary anymore with newer kernels. Unsure what exactly this could be but it seems odd to me that secure boot would be the culprit as all it does is prevent unsigned code from running during boot, but after that it should have no noticeable effect.