erpalma/throttled #9: Pip failed during installation process

f1xmAn, 2018-05-24:
install.sh script doesn't install dependencies via pip [(log)](https://pastebin.com/gtf5pzFj) on Fedora 28. I've installed all dependencies using dnf command and added —system-site-packages to virtualenv command in order to complete installation.

erpalma, 2018-05-25:
It seems that you are missing a package:

` Package x11 was not found in the pkg-config search path.`

Maybe you should install `libX11-devel`?

f1xmAn, 2018-05-25:
It's already installed.

erpalma, 2018-05-28:
I should run a Fedora VM and have a look at what's happening

f1xmAn, 2018-05-28:
I can provide you any additional information if needed.

f1xmAn, 2018-05-28:
Accidentally closed. Reopened :)

jdydsco, 2018-06-24:
Can someone please update the install guide to show how to use this with Fedora if it's possible, I've been having problems installing it as well. I found out what most of the packages were named since they're different on Fedora. It still says gobject-introspection not found even though I installed the only version I could find with dnf (gobject-introspection-1.56.1-1.fc28.x86_64)