erpalma/throttled #10: Just a question

boltex, 2018-06-07:
Sorry to use issues for this - i cant find how to message on github 
I'm new to linux in genereal and I'd like to know what this fix actually performs.

I tried to make sense of the code and installer ... Does this fix keep an instance of python running continually to set the values every couple of seconds? or does it kill it and restarts it ? or is it a cron job or something else? 
sorry if this is innapropriate... 
thank in advance

erpalma, 2018-06-07:
Yes, the script is kept running in background. Every `Update_Rate_s` the values for MSR and MCHBAR are re-applied. Dbus is also used to watch for the resume from suspend event in order to set undervolt values.

wintermute42, 2018-06-07:
Just a question (meaning I haven't investigated that myself yet): Would you
say that it is possible that the script running in the background causes my
X1 Carbon (Ubuntu 18.04) to wake up from suspend mode? I noticed that it
always wakes up without any action on my side after a couple of minutes.
This seems to be the case, no matter if I set acpi.ec_no_wakeup=1 or not.

wintermute42, 2018-06-07:
@boltex you mean "Config -> Thunderbolt BIOS Assist Mode - Set to "Enabled"? Yes, I did that. 
I also set _psmouse.synaptics_intertouch=1_ and removed _ i2c_i801_ from _/etc/modprobe.d/blacklist.conf_ to get the touchpad to work (which might not be necessary if you have a more recent kernel version than my 4.15 according to this https://medium.com/@maciejmensfeld/lenovo-thinkpad-x1-carbon-6th-gen-2018-ubuntu-18-04-tweaks-6e1d7758bd43)

erpalma, 2018-06-08:
@wintermute42 nice question. Are you on patched dsdt to enable proper S3 or still using Si03? I don't think that my script can actually cause any wakeup irq, but who knows. You can try to disable it (`systemctl stop lenovo_fix.service`) and report if wakeups still occur.

boltex, 2018-06-08:
@wintermute42 I am also curious about that

erpalma, 2018-06-09:
I think you should try to patch the DSDT and see what happens! It's not a permanent mod so no big worries

closed 2018-08-02 by erpalma