erpalma/throttled #11: Problem re-appeared after second boot

WallyYang, 2018-06-13:
There were several throttling warnings during boot before I installed the script and gone once after the installation. However, those warnings re-appeared after I boot my laptop again, any idea what might be the problem?

My laptop is T480 with i7-8650u
System: Arch Linux
Systemd version: 238

Thanks!

erpalma, 2018-06-14:
Do you mean these?

`CPU3: Package temperature above threshold, cpu clock throttled (total events = 164)`

Your CPU is throttling even with my script active, but it's doing that at a much higher threshold. Even with the script correctly working, the EC sometimes still reports those events to the kernel, I don't know exactly in which conditions but it happens to me too. You can check with [s-tui](https://github.com/amanusk/s-tui) if the script is working correctly though.

WallyYang, 2018-06-14:
Yep, I can confirm it's working correctly, I guess Lenovo really failed this generation of ThinkPad. Thank you for your script, I shall close this issue.

closed 2018-06-14 by WallyYang