erpalma/throttled #12: Works for X1C6

mensfeld, 2018-06-19:
Hey, just wanted to confirm that it works for X1 C6 and I've used it for my tutorial: https://mensfeld.pl/2018/05/lenovo-thinkpad-x1-carbon-6th-gen-2018-ubuntu-18-04-tweaks/

Undervolting seems to work as well (so will update the art with that as well).

erpalma, 2018-06-19:
Mentioned your page in the readme ;)

mensfeld, 2018-06-19:
Thank you. Thank you also for this amazing piece of work. It took me a fair amount of time to get my x1c6 going and being able to tackle the throttling was one of the most important parts of my fight :)

closed 2018-06-19 by mensfeld

lakshmi-kannan, 2019-04-09:
FWIW, I've a X1 carbon 6th gen and the undervolting results in hard freeze. I undervolted only for battery. 

## CPU

```
vendor_id	: GenuineIntel
cpu family	: 6
model		: 142
model name	: Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
stepping	: 10
microcode	: 0x9a
```

## OS
Fedora 30

I used the undervolting settings from https://medium.com/@hkdb/ubuntu-18-04-on-lenovo-x1-carbon-6g-d99d5667d4d5 which appears same as the one @mensfeld linked to https://mensfeld.pl/2018/05/lenovo-thinkpad-x1-carbon-6th-gen-2018-ubuntu-18-04-tweaks/. Other than undervolting, the throttling fix seems to work? Would be good to document validation steps. Also, the README recommends that the throttling fix be enabled via systemctl enable which is dangerous IMO. I did that and my laptop kept freezing. Luckily, I remembered that connecting to power would disable undervolting from kicking in and reset it back to all 0s for battery mode. I do recognize that making the change and testing it is the only option but not recommending systemctl enable until people verify that there are no freezes will save them from some frustration.

erpalma, 2019-04-09:
As stated in the README you should find the undervolt values that work on **your** laptop, not just copy/pasting values found online. Again, you can check if the tool is working by using `--debug` and `--monitor` options and/or s-tui.

mensfeld, 2019-04-09:
It did work for me and the second X1 that I had in my hands. I no longer own this laptop so hard to check again sorry :(