erpalma/throttled #15: Expand support / documentation for other laptops

galfwender, 2018-06-30:
Thanks for your work on this.  I just wanted to let you know that it also works on laptops from other manufacturers, e.g. Dell XPS 9370 with the same cpu.  Perhaps the documentation could be updated to include this?  I am also interested to know if the undervolting would work (I felt it best not to try it)?

On my XPS it now manages to sustain between 2.8 and 3.0 Ghz with temp between 97-99C depending on ambient.  It no longer scales back to 2.2-2.4 Ghz with temps of 80C.

However I did note that the service fails if a temp of more than 97C is specified.  In my case, although pushing the limits, it seems to be OK with the value manually set at 98 using wrmsr and it eeks out a further 50-100Mhz.

galfwender, 2018-07-01:
Perhaps instead of silently failing, the script could cap the limit at 97 instead?  I was puzzled for a few minutes as to why it was not working.

erpalma, 2018-07-01:
Good advice thank you! I've just updated the config loading code.

closed 2018-07-05 by galfwender