erpalma/throttled #16: KDE Neon broken completely after trying this fix

achlubek, 2018-07-01:
System was stable and I ran some benchmark, then I rebooted, still stable, Suddenly GPU started to show artifacts in very specific scenarios (some shaders in opengl) and then system became unstable, lagging for seconds every 30 seconds or so. I tried editing the config but filesystem became readonly, I forced a reboot and the system is now unbootable completely, shows only initramfs sheel at boot. Well I guess I should have made a backup :)

achlubek, 2018-07-01:
Running recovery kernel showed more info, particularly which partition was corrupted, and suggested running fsck, after running it against the partition the system booted. I disabled the fix just after the boot.

erpalma, 2018-07-02:
Can you point out on which notebook you tested the script? By GPU you mean the integrated one? Did you applied undervolt or not? Can you post `lenovo_fix.conf`?

achlubek, 2018-07-02:
Ah I forgot to add I did the undervolting but not that much as in readme, there is what I did, other parameters were left as default:
`[UNDERVOLT]
# CPU core voltage offset (mV)
CORE: -70
# Integrated GPU voltage offset (mV)
GPU: -70
# CPU cache voltage offset (mV)
CACHE: -70
# System Agent voltage offset (mV)
UNCORE: -70
# Analog I/O voltage offset (mV)
ANALOGIO: 0`
The laptop is T480 with 8250u and integrated GPU

erpalma, 2018-07-02:
That's the issue. Your CPU is not stable with those values and crashed badly. You should find the right values for your system by stress testing and slowly raising (lowering since are negative) those values until your system becomes unstable. Then you step back a little (like 20mV) and you *should* be ok. 

I'll add a warning about this to the doc.

achlubek, 2018-07-02:
I will try that, thank you

closed 2018-07-02 by achlubek