erpalma/throttled #18: crash after stop stress immediately

chopin1998, 2018-07-10:
Hi, thanks very much for your code.

i have a question, i setup my t480s, use ~ -90mv undervoltage, and ~92oC temp wall.

it can run stress(in s-tui) very long time, but crash after stress stop immediately, i can watch CPU freq down from 3.xG to 800Mhz in seconds, then system crash..

some suggestion?

erpalma, 2018-07-10:
Yep, that's quite normal. The crash on transition from high freq + full load to low freq is a symptom that the undervolt is too high (i.e. too low voltage). Try go with -70/-80 mV. Also which voltage plane are you talking of?

chopin1998, 2018-08-04:
[UNDERVOLT]
# CPU core voltage offset (mV)
CORE: -105
# Integrated GPU voltage offset (mV)
GPU: -100
# CPU cache voltage offset (mV)
CACHE: -90
# System Agent voltage offset (mV)
UNCORE: -100
# Analog I/O voltage offset (mV)
ANALOGIO: 0

now, i use those config, that's fine.  Thank you!

closed 2018-08-30 by DEvil0000