erpalma/throttled #19: Why isn't it event driven?

Venemo, 2018-07-19:
Instead of writing to the registers every X seconds, I believe it should be enough to just write to them at specific events, ie. after boot and after waking from sleep. Am I missing something?

mimecry, 2018-07-19:
the embedded controllers in the new gen of Thinkpads reset these values during usage (i don't know the specific interval), so you need to be constantly writing these values to get them to stick

Venemo, 2018-07-19:
Wow. I don't see why they do that, but in that case I understand. I'm just using this on a Dell XPS 9370, and the controller here doesn't seem to be as evil.

By the way AFAIU your stuff should be safely usable on all Skylake and newer Intel CPUs, not only Lenovo laptops.

erpalma, 2018-07-19:
Yes, it should work on any recent Intel laptop. The EC resets some values at some unknown events so, since I can't be notified about these events I just override the registers at fixed intervals. I agree that it is not optimal/clean/beautiful but it's the only solution I've found. In the end the script spend a couple of milliseconds of cpu time every X seconds, so it's not that bad.

Venemo, 2018-07-24:
Would be interesting to try to research what events trigger the EC to mess with these registers.