erpalma/throttled #20: Is 44W necessary?

Venemo, 2018-07-19:
Where does the 44W default come from? If I read your posts correctly you said you have a 8550U, but according to [the 8550U Intel website](https://ark.intel.com/products/122589/Intel-Core-i7-8550U-Processor-8M-Cache-up-to-4_00-GHz) its maximum TDP-up is 25W. Or am I misreading it?

erpalma, 2018-07-19:
25W is the default TDP-up yes, but it can be further raised if the cooling system is adequate. On Windows the maximum TDP on AC is 44W. Clearly notebooks of this size cannot sustain this power level for long periods, but for short bursts it's ok. So for short bursts you are limited by the 44W value and then, for longer runs, the limits becomes the trip temperature.

Venemo, 2018-07-21:
Good to know, thank you!

closed 2018-07-21 by Venemo