erpalma/throttled #21: PermissionError when trying to write to voltage MSRs

stevekerrison, 2018-07-20:
This great little program doesn't work in its current version on Ubuntu 18.04 and T480. It seems the script is unable to write voltage configuration.

I'm not interested in undervolt so I commented out the [call to `undervolt`](
https://github.com/erpalma/lenovo-throttling-fix/blob/56a4c5430c29d41b16aee1d5c69de75fc417c551/lenovo_fix.py#L186), so the rest of the script works fine.

Relevant log below, kernel is `4.15.0-23-generic`.

```
ul 19 18:32:05 steve-ThinkPad-T480 kernel: [ 2115.556629] msr: Direct access to MSR 150
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]: Traceback (most recent call last):
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:   File "/opt/lenovo_fix/lenovo_fix.py", line 226, in <module>
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:     main()
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:   File "/opt/lenovo_fix/lenovo_fix.py", line 186, in main
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]: Exception in thread Thread-1:
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]: Traceback (most recent call last):
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:   File "/opt/lenovo_fix/mmio.py", line 58, in _open
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:     fd = os.open("/dev/mem", os.O_RDWR | os.O_SYNC)
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]: PermissionError: [Errno 1] Operation not permitted: '/dev/mem'
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]: During handling of the above exception, another exception occurred:
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]: Traceback (most recent call last):
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:   File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:     self.run()
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:   File "/usr/lib/python3.6/threading.py", line 864, in run
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:     self._target(*self._args, **self._kwargs)
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:   File "/opt/lenovo_fix/lenovo_fix.py", line 143, in power_thread
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:     mchbar_mmio = MMIO(0xfed159a0, 8)
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:   File "/opt/lenovo_fix/mmio.py", line 33, in __init__
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:     self._open(physaddr, size)
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:   File "/opt/lenovo_fix/mmio.py", line 60, in _open
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:     raise MMIOError(e.errno, "Opening /dev/mem: " + e.strerror)
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]: mmio.MMIOError: [Errno 1] Opening /dev/mem: Operation not permitted
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:     undervolt(config)
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:   File "/opt/lenovo_fix/lenovo_fix.py", line 67, in undervolt
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:     writemsr(0x150, calc_undervolt_msr(plane, config.getfloat('UNDERVOLT', plane)))
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:   File "/opt/lenovo_fix/lenovo_fix.py", line 43, in writemsr
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]:     os.write(f, struct.pack('Q', val))
Jul 19 18:32:05 steve-ThinkPad-T480 python3[23214]: PermissionError: [Errno 1] Operation not permitted
Jul 19 18:32:05 steve-ThinkPad-T480 systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=1/FAILURE
Jul 19 18:32:05 steve-ThinkPad-T480 systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.
```

stevekerrison, 2018-07-20:
I see this is mentioned in #6 , but perhaps some extra checks would avoid a crash.

erpalma, 2018-07-20:
I've just add some more checks as suggested.