erpalma/throttled #24: the script improves the situation but there is something limiting still L480 with i7-8550U

DEvil0000, 2018-08-01:
I have an L480 with a i7-8550U. (latest bios)
The script improves the situation but there is still something throtteling the CPU without need and limiting the potential. When I start stress with c of 1,2 or 8 I first get a period of high CPU performance (like I would expect it) of about 14 to 28 secodns. Then it gos with the following pattern: it drops very drastically for about 3 seconds followed by a higher (but not as high as at the beginning) utilisation for about 3 seconds. This pattern then repeats as long as cpu stress is there. Even if the temperature is 10°C below the thermal threshold (stress with c of 1 or 2) the performance goes down every 3 seconds for 3 seconds. When it goes down the temperature drops imeadiately to about 55 to 63°C. As I understand it there should be no need to throttle that long, much and offten. I guess there is some other magic register in the CPU involved.
I can see similar patterns on the machines of colleges using the s version.

Beside that lenovo does let the fan only run 2/3 of its max speed (which gets quit loud). Thinkfan solves this for me but the effect on performance is low anyway. Seems like lenovo did not design the laptops properly to get the heat away.

Let me know if I should send you some MSR values.

erpalma, 2018-08-01:
Are you using TLP? If yes, can you post your config (on pastebin)? Also try to set `Update_Rate_s` to 1.

DEvil0000, 2018-08-02:
I have tlp installed and configured to load the battery only to 60%. I do not use other aspects of it intentionally. I did disable it by config now for testing (/etc/default/tlp) and did stop the tlp service. I also tried lenovo throttle fix config with 1second re-apply. All those things do not change the behaviour.

DEvil0000, 2018-08-02:
I also found MSR 0x1B0 set to 0110b = 6 and did try setting it to 0. It does not seem to have a big effect but I think the CPU is using turbo more early with it set to 0.

erpalma, 2018-08-02:
Can you please test again with TLP enabled also for HWP and x86_energy_perf_policy? Or you can manually set both to performance.

DEvil0000, 2018-08-02:
Another thing seems to be MSR 0x64b.
It was set to 0 and the drops did go down to about 12W.
Setting it to 1 did the drops go down to about 7W.
Setting  to 2 did the drops only go down to 14/15W.
So it seems the script should set this to 2, too.

DEvil0000, 2018-08-02:
TLP enabled or disabled does not seem to have a effect. The described issue is the same with or without it.
TLP config:
```
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
TLP_PERSISTENT_DEFAULT=0
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_HWP_ON_AC=performance
CPU_HWP_ON_BAT=balance_power
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=power
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
AHCI_RUNTIME_PM_TIMEOUT=15
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_AC=0
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
USB_AUTOSUSPEND=1
USB_BLACKLIST_BTUSB=0
USB_BLACKLIST_PHONE=0
USB_BLACKLIST_PRINTER=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0
START_CHARGE_THRESH_BAT0=45
STOP_CHARGE_THRESH_BAT0=75
```
lenovo throttling fix config:
```
Enabled: True
Update_Rate_s: 1
PL1_Tdp_W: 29
PL1_Duration_s: 28
PL2_Tdp_W: 44
PL2_Duration_S: 0.002
Trip_Temp_C: 85
Update_Rate_s: 1
PL1_Tdp_W: 44
PL1_Duration_s: 28
PL2_Tdp_W: 44
PL2_Duration_S: 0.002
Trip_Temp_C: 95
HWP_Mode: True
CORE: 0
GPU: 0
CACHE: 0
UNCORE: 0
ANALOGIO: 0
```
I did not know of x86_energy_perf_policy and installed it.
However the output is the same with TLP running or not.
x86_energy_perf_policy output:
```
cpu0: EPB 0
cpu0: HWP_REQ: min 4 max 40 des 0 epp 128 window 0x0 (0*10^0us) use_pkg 0
cpu0: HWP_CAP: low 1 eff 8 guar 20 high 40
cpu1: EPB 0
cpu1: HWP_REQ: min 4 max 40 des 0 epp 128 window 0x0 (0*10^0us) use_pkg 0
cpu1: HWP_CAP: low 1 eff 8 guar 20 high 40
cpu2: EPB 0
cpu2: HWP_REQ: min 4 max 40 des 0 epp 128 window 0x0 (0*10^0us) use_pkg 0
cpu2: HWP_CAP: low 1 eff 8 guar 20 high 40
cpu3: EPB 0
cpu3: HWP_REQ: min 4 max 40 des 0 epp 128 window 0x0 (0*10^0us) use_pkg 0
cpu3: HWP_CAP: low 1 eff 8 guar 20 high 40
cpu4: EPB 0
cpu4: HWP_REQ: min 4 max 40 des 0 epp 128 window 0x0 (0*10^0us) use_pkg 0
cpu4: HWP_CAP: low 1 eff 8 guar 20 high 40
cpu5: EPB 0
cpu5: HWP_REQ: min 4 max 40 des 0 epp 128 window 0x0 (0*10^0us) use_pkg 0
cpu5: HWP_CAP: low 1 eff 8 guar 20 high 40
cpu6: EPB 0
cpu6: HWP_REQ: min 4 max 40 des 0 epp 128 window 0x0 (0*10^0us) use_pkg 0
cpu6: HWP_CAP: low 1 eff 8 guar 20 high 40
cpu7: EPB 0
cpu7: HWP_REQ: min 4 max 40 des 0 epp 128 window 0x0 (0*10^0us) use_pkg 0
cpu7: HWP_CAP: low 1 eff 8 guar 20 high 40
```

DEvil0000, 2018-08-02:
We should be able to write to MSR 0x614 as far as I understand but I can't write to it with wrmsr.  I can read however. This might be something interesing as well. Any idea why I can not write to it?

DEvil0000, 2018-08-03:
Some more findings:
It looks like the throttle kicks in when the CPU gets over 80°C for 3 seconds. I let the script apply a limit of 95°C which results in a MSR 0x1a2 Temperature Target=100 and  Target Offset=5. This sounds correct to me and works for the first ~14-28 seconds when the CPU gets some load but later it does not seem to work correct.
Is it possible to change the hidden 3s window or 80°C throttle?
Is there something else that should get set? with ACPI maybe?

In some cases I also see the Max Turbo Limit Log kicking in. This makes me think that the turbo ratio per core is something we should try to set.

I figured out that my Lenovo L480 is able to keep up with constant 17W to 18W on 8 cores in my about 23°C office room. This seems to be the limit of heat dissipation this device can handle with ~76°C - not triggering the 80°C throttle. However with less cores running some turbo it gets hotter with less Watts. For example with 2Cores getting load the temperature goes to ~85°C which seems to cause the throttle every 3 seconds for 3 seconds later.

erpalma, 2018-08-03:
Interesting. We need to investigate more. Maybe the EC is rewriting those values more frequently than in my T480s? I'm sorry but I don't think I could help for a couple of weeks since I'll be very busy :/

If you want you can keep looking for the cause of this issue and report here, it would be very helpful.

DEvil0000, 2018-08-03:
**Finally I figured it out !!**
**_Thermald_** is installed (ubuntu default as it seems) and does the throttle over setting _RAPL_ !
It is reading temperatures for the thermal zones and setting trip points for slowing down. It was setting a _80°C_ trip on `/sys/class/thermal/thermal_zone2/` and `zone3` which are of type `B0D4` and `SEN1`. I am not sure for which sensors those zone types stand for but hopefully CPU. However - since the limit is set quite low (lower then my 95°C in lenovo throttle fix config) thermald starts messing with the CPU RAPL, P-states, frequencies and everything settings and slowing stuff down. Sadly it is slowing it down quite dramatic - way more then ever needed. There is no need at all for thermald as far as I understand it - at least for the modern CPUs. The config is (quite useless) default from the package which was installed on ubuntu install as it seems. Stopping the service solves all issues.

Open Questions:
Why is thermald installed?
Why is it slowing the CPU down that much?
Does it make sense to configure it and let it run (I think: no)?

Feature request:
* Please add more options to the tool - like setting MSR 0x1B0 (switching profile) and MSR 0x64b (cTDP up if I am correct). Also setting turbo ratios there would be ncie. Should I open a new feature request issue for it?
* you can add the L480 to the list + a FAQ hint to disable thermald

Performance on the L480:
Core undervolted with -90 (rest is 0 atm) and 0x64b set as described above
You can run the cpu fan on 100% with thinkfan (which it would not do normally) but this is getting quite load and will allow you about 1W more then listed below.
1Core -> 3,6GHz 16W 90°C (don't know why the turbo is not kicking it higher)
2Cores -> 3,6GHz 25W 95°C (tiny throttle after a while to 22W)
3Cores -> 3,2GHz 25.5W 95°C
8Cores -> 2,9GHz 28W 95°C (after a while it is throttleing down a bit to 27W sometimes 26W)
TLP did not doing something to it here.

erpalma, 2018-08-03:
Uh nice finding! I'm on Gentoo so I have no packages that I didn't installed ;) But yes, I really forgot the existence of thermald... And I don't think you actually need it.

Yes please, add a new feature request so we can discuss about that and track the changes. I'll give a look to what you suggested but I think I can integrate those registers in the settings.

closed 2018-08-03 by DEvil0000

wintermute42, 2018-08-04:
do you know whether it is safe to uninstall thermald on Ubuntu 18.04? if you did, were there any unexpected side effects? or do you have a recommendation how exactly to disable it permanently?

whnr, 2018-08-04:
Just a quick comment: X1C6, with i7 8550u and Arch Linux: With TLP enabled on BAT there seems to ba a hard frequency limit. With TLP off the CPU is only thermally limited.

erpalma, 2018-08-04:
Well on battery that's quite the job of TLP. You can force `tlp ac` or configure some of the settings in the battery profile to be more aggressive on power (at the cost of battery life clearly). Check `CPU_HWP_ON_BAT` and `ENERGY_PERF_POLICY_ON_BAT`.

wintermute42, 2018-08-05:
> @wintermute42 <https://github.com/wintermute42> I do not see a need to
> have thermald at all for modern CPUs. Your laptop will run more hot then
> before but this is what we want to get the CPU power. Having thermald might
> be important for some older CPUs without proper thermal throttle (TCC for
> intel) or thermal throttle disabled. It might also be usefull on a laptop
> if configured correctly. However before turning thermald off you should
> make sure your CPU has TCC activated since I am not sure if it will
> throttle due to heat otherwise. I don't know why it should ever be off but
> who knows. To check it you can run sudo rdmsr -d 0x1a0 -u -f 3:3 - if it
> returns a 1 you are fine. Without TCC beeing enabled the temperature target
> set by the script is ignored anyway - I think.
> So I did not uninstall it till now i just disabled it by systemctl
> disable thermald.service (you should also stop it).
>
> @whnr <https://github.com/whnr> you can change the configuration of TLP
> or turn it off but most people like to get more battery hours when they
> travel instead of high performance.

Great, thank you for the explanation. I guess TCC should be activated by
default (talking about a X1C 2108 in my case), but I will check to be on
the sure side. Do you maybe know whether this corresponds to the setting
"Adaptive Thermal Management" in the Thinkpdad BIOS?

DEvil0000, 2018-08-06:
Sorry I can not tell you what this option is. TCC should be on by default and always on from my point of view. This option however could be related or not.

wintermute42, 2018-08-06:
Ok, thank you