erpalma/throttled #25: Max boost and fall back on T480 with i7-8650U

wittich, 2018-08-02:
Hi there,

thx so much for the fix it helped me to avoid to run into a total throttling where the CPU get stuck with 0.2 Ghz until a reboot.

Two particular issues I saw now with the default config of the `lenovo-throttling-fix`. 

1. The boost seems so be set on default to 4.0 Ghz which I can monitor, but the [specs](https://ark.intel.com/products/124968/Intel-Core-i7-8650U-Processor-8M-Cache-up-to-4_20-GHz) mention a max tubo of 4.2 Ghz.
2. when enabled the fix the CPU seems slightly slow on longer workload even though the boost is there. I have the feeling that the base frequencies are maybe different to the default settings of Lenevo.

Any comments on that or how I can measure or monitor my observation to have some objective evidence?

Regards
Valentin

erpalma, 2018-08-02:
Have you tried using `s-tui` tool to monitor frequencies/power/temps? 

 1. My script is not overriding the max turbo multiplier. How do you know that your cpu is limited to 4 GHz turbo? Have you tried enabling the HWP option?
 2. You are saying that the cpu is slow even if frequencies are in the boost region? BTW I'm not messing up with any cpu multiplier so the script is not altering the base and turbo frequencies. What it does is to "allow" the cpu to clock as high as your cooling system can handle.

DEvil0000, 2018-08-03:
Is thermald running on your machine? something else maybe messing with the same settings? (see #24)

My i7-8**5**50U on the L480 is also not using 100% of the turbo for unclear reasons. I guess a turbo ratio is set too low by default. However for 2 or more cores processing its now running quite fast with up to 28W.

Can you provide more details on the behavior? stress for 1 core, 2 cores, 8 cores? screenshot of s-tui when running stress for lets say 60 seconds?

halbleben, 2018-08-04:
I have a similar problem on the X1 Carbon 6th with the i7-8650U. It oscillates with a period of about 20s between the configured maximum temperature (in my case 85C) to about 74C. 
Here is a screenshot of how that looks like in s-tui:
[image]

The same happens at a max temp of 95C. Using HWP_Mode: True does not change that.

Here is my conf:
[lenovo_fix.txt](https://github.com/erpalma/lenovo-throttling-fix/files/2259458/lenovo_fix.txt)

halbleben, 2018-08-04:
I'm not sure whether this is the right place to share this, but I just wanted to share my results for the X1 Carbon 6th with i7-8650U because it's quite hard to find numbers for this CPU on the web.

Basics:
Trip_Temp_C: 95
HWP_Mode: True

[UNDERVOLT] (minimum -> reducing all voltages by 10mV makes it unstable)
CORE: -90
GPU: -50
CACHE: -90
UNCORE: -75
ANALOGIO: 0

RESULT (with around 33C ambient temperature, about 15min runtime)
8 cores (s-tui stress test):  3,2 GHz, 23W
1 core (prime95): 3,9 GHz, 18W

Using thinkfan and running the fan at 100% only changes these results marginally, altering it by at most 0.5W and 50MHz. I guess it's not worth the noise.

Again, crucially, thermald has to be disabled to avoid those weird oscillations reported in #24 and by me in above post.

DEvil0000, 2018-08-04:
Quite interestingly you had a similar pattern also caused by thermald but with a much longer period.

DEvil0000, 2018-08-30:
thermald was the issue. see above

closed 2018-08-30 by DEvil0000