erpalma/throttled #26: [improvement] make cTDP configurable

DEvil0000, 2018-08-03:
If I am correct this is MSR 0x64b.
0 is normal.
1 is the low profile.
2 is the high profile.

see my comment in [my other ticket](https://github.com/erpalma/lenovo-throttling-fix/issues/24#issuecomment-409851402)

DEvil0000, 2018-08-03:
[This](https://github.com/DEvil0000/lenovo-throttling-fix/tree/improvement/configure_cTDP) might work - can test it on monday.

erpalma, 2018-08-04:
I guess we should document this new feature well on the readme, which is getting quite long btw! :)

wintermute42, 2018-08-05:
I just want to mention that this script turned out to be a pretty great
solution and that I, as a user of the script, really appreciate how well it
is documented in the readme :-)

closed 2018-08-13 by DEvil0000