erpalma/throttled #29: Is there any way to check if undervolt was successful?

PCK1992, 2018-08-06:
I couldn't find a way to verify whether my CPU has been undervolted or not . Any way to test if the package works completely?

DEvil0000, 2018-08-13:
Not in a user friendly way but its working (for me).

erpalma, 2018-08-13:
We are going to add read back and check functions for the registers value. Besides that you can stress test your pc and see if it can clock higher under full load and thermal limit.

sanderboom, 2018-08-17:
Related: is there any way to check if Power Throttling is active?

erpalma, 2018-08-17:
I've add the `--debug` option, you might want to check it.

About Power/Thermal/Current Throttling I can add a `--monitor` option or so.

sanderboom, 2018-08-18:
Thanks for the `--debug` option!

```
[E] No root no party. Try again with sudo.
```
:-)

One suggestion would be to add the voltage (in decimals) to the output. 

A second suggestion would be to be able to follow the debug output using `journalctl -f`. I've added `--debug` to `/usr/lib/systemd/system/lenovo_fix.service` wondering if the output would show up, but this is not the case. 

> About Power/Thermal/Current Throttling I can add a --monitor option or so.

That would be great! As I (and many of us I guess) are spoiled with the great overview that [XTU](https://downloadcenter.intel.com/download/24075/Intel-Extreme-Tuning-Utility-Intel-XTU-) provides, it would be great to have a way to monitor types of Throttling on Linux.

erpalma, 2018-08-18:
I'll add more debugging output and the monitoring feature in the following days.

Meanwhile I've updated the systemd service file by adding 
```
StandardOutput=syslog
StandardError=syslog
```
so you can track the --debug output via `journalctl` as requested.

sanderboom, 2018-08-22:
Thank you for implementing this!

The strangest thing is happening on my installation though: I only have `lenovo_fix` entries via `journalctl` on the day that I installed this update (2018-08-18). After a reboot `lenovo_fix` messages stopped appearing. I will investigate this myself, although I wanted to mention this in case someone else runs into this 'curiosity'. I'm now running version: `0.75.bbef7bf-1`.

mcginty, 2018-08-29:
The [undervolt](https://github.com/georgewhewell/undervolt) python tool has a `--read` mode that will show you the current voltage configuration.

MartinX3, 2018-08-29:
Thank you.
Now I can see, that the -150mv are applied.

erpalma, 2018-08-30:
Now `--debug` shows the offset write/read values in mV too.

closed 2018-08-30 by erpalma

sanderboom, 2018-09-01:
One more question: do you have any idea why `lenovo_fix` debug messages do not show up using `journalctl`? Just curious if it works for you. Could be something wrong with my system, but didn't find a clue yet. TIA.

```
λ journalctl -f -u lenovo_fix
-- Logs begin at Tue 2018-07-24 22:06:50 CEST. --
aug 27 07:46:13 t480s systemd[1]: Started Stop Intel throttling.
aug 27 11:17:59 t480s systemd[1]: Stopping Stop Intel throttling...
aug 27 11:17:59 t480s systemd[1]: Stopped Stop Intel throttling.
-- Reboot --
aug 27 11:18:22 t480s systemd[1]: Started Stop Intel throttling.
aug 28 23:10:09 t480s systemd[1]: Stopping Stop Intel throttling...
aug 28 23:10:09 t480s systemd[1]: Stopped Stop Intel throttling.
-- Reboot --
aug 28 23:27:05 t480s systemd[1]: Started Stop Intel throttling.
sep 01 10:21:01 t480s systemd[1]: Stopping Stop Intel throttling...
sep 01 10:21:01 t480s systemd[1]: Stopped Stop Intel throttling.
sep 01 10:25:53 t480s systemd[1]: Started Stop Intel throttling.
```

erpalma, 2018-09-01:
Do you have
```
StandardOutput=syslog
StandardError=syslog
```
and `--debug` in your `.service` file?

sanderboom, 2018-09-01:
Works. I was under the impression that `--debug` was not need in the `.service` file. Thanks for clearing this up.

sanderboom, 2018-09-04:
> About Power/Thermal/Current Throttling I can add a --monitor option or so.

I've also raised the matter at the `s-tui` [repository](https://github.com/amanusk/s-tui/issues/83#issuecomment-418176360). Not sure which tool would be more suitable. 

Seeing discussion at https://github.com/erpalma/lenovo-throttling-fix/issues/46 about the scope and name of `lenovo-throttling-fix`, maybe this could be the start of a cooperation between two excellent tools? :)