erpalma/throttled #33: Undervolting Linux vs Windows, AC vs Battery

sanderboom, 2018-08-17:
First of all: thanks for the great work and the awesome tool! Filling one of the gaps that the multi million dollar company that claims to be linux friendly should not have left open ;) 

First question is regarding undervolting differences between windows and linux: is it normal to find a (big) difference in possible undervolting settings between the two OS's? In windows I can run prime95 a day with -120mV on core and cache. In linux (using this script) -100mV is unstable (many firefox crashes). Theoretically this should not be the case I assume? How is your experience regarding this difference? 

Second question is regarding undervolting differences between running plugged-in vs on battery. I found that I can run stable with a certain undervolt setting on AC, but this crashes the desktop (regardless the OS) when running on battery. I assume this has to do with the lowering of the default voltage bsy energy saving processes when on battery. The threshold is now too low and the system becomes unstable. Does this make sense? And in this case: would it be an idea to add different undervolting settings for plugged-in and on battery?

Thanks.

erpalma, 2018-08-17:
The Windows vs Linux difference you are experiencing is probably due to the different kernel architecture of the two OS. In general I found Linux to be far more sensitive to computation/memory errors induces by undervolt or overclock. I guess it's a good thing since the errors are still occurring on Windows but you are just not detecting them. 

An interesting and unexpected fact about undervolting is that, often, the lower/idle frequencies are much less tolerant to low voltages than full load. You might get a rock-solid system while running prime but when you stop it, the transition from high to low frequency cause a freeze. I don't think the CPU is lowering its voltage just because it's on battery, it would not make sense. The CPU is always using the lowest stable voltage set by Intel. The problem is that we are applying a voltage offset and the CPU sets its voltage depending on the actual frequency. It might be the case that a voltage offset is good for top frequencies but it's too high for the lowest ones since the voltage at idle is already quite low.

DEvil0000, 2018-08-17:
you are right the first issues normally occur when switching frequencies. like turbo or even normal operation would do but prime would not. however starting and stopping a 50% to 100% load for 1,2,3,4,8 threads would.
Beside that from my understanding having ac or battery powering could make a (tiny) difference. 
If it crashes in whatever situation you undervolted too much. You just can run stable with high frequencies and a lot of undervolting. The effect on heat for undervolting is quite tiny anyway and if you undervolt too much it might do retransmissions on some bus (which i don't know) and you do not get the desiered effect at all.
For me -100 or even -90 was the limit.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/erpalma/lenovo-throttling-fix/issues/33#issuecomment-413906900",
"url": "https://github.com/erpalma/lenovo-throttling-fix/issues/33#issuecomment-413906900",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [erpalma/lenovo-throttling-fix] Undervolting Linux vs Windows, AC vs Battery (#33)",
"sections": [
{
"text": "",
"activityTitle": "**Francesco Palmarini**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@erpalma",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"erpalma/lenovo-throttling-fix\",\n\"issueId\": 33,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"erpalma/lenovo-throttling-fix\",\n\"issueId\": 33\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/erpalma/lenovo-throttling-fix/issues/33#issuecomment-413906900"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 369473328\n}"
}
],
"themeColor": "26292E"
}
]

sanderboom, 2018-08-20:
Thanks for the thorough replies. 

I had the idea that the voltage would be influenced by the powersaving measurements of the OS while running on battery. But as I understand correctly there is no change to the voltage's, but it's the lower overall frequencies that trigger the crashes I experience when undervolting when running on battery.

DEvil0000, 2018-08-20:
Its the change from one to some other frequency. Also known as changeing the multiplier (per core). This is something the CPU does all the time in normal operation when the load is changeing.

"activityTitle": "**Sander Boom**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@sanderboom",

closed 2018-08-30 by DEvil0000