erpalma/throttled #36: T460P support planned?

MartinX3, 2018-08-20:
I own a T460P with i7-6700HQ 3.5 GHz CPU and a GTX 940MX GPU.
Using this tool to undervolt my cpu and its cache by -150mV lowers the max temp from 90 to 80°C
But the throttling issue still exists.

Even if only one core reaches 100% while the others are doing nothing the throttling will be activated.
But it won't be deactivated after the ThinkPad is getting colder.
Means my OS lags and stutters every 2-3 Seconds (Video/Mouse Movement/Windows/Animations).
But not the sound.

PS:
It is weird that the max clock is 3.2GHz on only one Core instead of 3.5GHz turbo.
The s-tui tool is telling me, that power limit is 28 Watt without throttling and 25, 20 or 15 Watt after throttling.

DEvil0000, 2018-08-20:
This tool is written for i7-8xxx CPUs (like i7-8550u) your CPU is 6th generation. Its not written for older CPUs of older laptops. However some things should also work with 6th gen CPUs (and maybe even with older ones). Please have a look at your BIOS options first and check if your cooling can work fine (e.g. no dust in the fan or in the way of the air).
Your CPU has only cTDP down but not up - so I am not sure if setting cTDP will work correct. The processors normal TDP is 45W depending on some things e.g. your cooling it should reach it and may stay there.
Setting a Temperature target may or may not work for this CPU - I am not sure here.
-150mV is a lot and may be too much for working correct and stable.
Beside that its quite normal that you basically never reach max turbo even if you stress only one CPU.
Please post the result for `rdmsr -d 0xce`

@erpalma should we document more explicit that this tool is written for i7-8xxx CPUs and check the CPU type in the script? or do you plan adding support for older ones too?

DEvil0000, 2018-08-20:
Looks like the CPU has cTDP with 1 down profile but no up profile beside default - as said before (you should use 0 in the config). It seems to have configurable temperature target (you can set it to 95 in the script) and configurable turbo multiplier (not supported by the script yet but normally set to max anyway). Beside that you can set voltage offsets (not sure since which CPU architecture) and HWP (starting with skylake). I am not sure if setting power limits is supported but thats not your issue anyway as it seems.
You can use `--debug` option to check if the values get written (which does not mean they are applied).

Please execute `wrmsr 0x19c 0` before a throttle happens and post the result of `rdmsr -d 0x19c` after the throttle happened.

A shutdown (not a reboot) will reset your CPU registers and BIOS may change them on any boot.

MartinX3, 2018-08-20:
Thanks in advance for your help!

My system is stable at -150mV under a stress test on all 8 cpu threads, while watching a video.

Every 2 Seconds my system is stopping for max 1 second.

So, the result of
rdmsr -d 0x19c

```
2284716034
```

rdmsr -d 0x19b

```
19
```

MartinX3, 2018-08-21:
Thank you for your feedback :)

According to the installation manual, I disabled thermald

The temperature throttling is fine.
That's not the problem.
The problem is, that I need to restart the ThinkPad to stop the throttling.
It don't stop to lag every 2 seconds, even if the cpu cools down and has no workload.

erpalma, 2018-08-21:
Did you check with `--debug` if, after throttling, the registers are written correctly? It might be the case that the EC/BIOS is locking one of the registers only after throttling has occurred.

MartinX3, 2018-08-22:
While my system is throttling and lagging, I get this output:

```
[W] cTDP setting not supported by this CPU
[W] cTDP setting not supported by this CPU
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match True
[D] Undervolt plane CORE - write 0xecc00000 - read 0xecc00000 - match True
[D] MSR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] Undervolt plane GPU - write 0x0 - read 0x0 - match True
[D] Undervolt plane CACHE - write 0xecc00000 - read 0xecc00000 - match True
[D] Undervolt plane UNCORE - write 0x0 - read 0x0 - match True
[D] Undervolt plane ANALOGIO - write 0x0 - read 0x0 - match True
[D] HWP for cpu0 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu1 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu2 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu3 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu4 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu5 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu6 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu7 - write "balance_performance" - read "balance_performance" - match True
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match True
[D] MSR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] HWP for cpu0 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu1 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu2 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu3 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu4 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu5 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu6 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu7 - write "balance_performance" - read "balance_performance" - match True
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match True
[D] MSR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] HWP for cpu0 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu1 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu2 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu3 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu4 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu5 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu6 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu7 - write "balance_performance" - read "balance_performance" - match True
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match True
[D] MSR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] HWP for cpu0 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu1 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu2 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu3 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu4 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu5 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu6 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu7 - write "balance_performance" - read "balance_performance" - match True
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match True
[D] MSR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] HWP for cpu0 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu1 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu2 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu3 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu4 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu5 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu6 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu7 - write "balance_performance" - read "balance_performance" - match True
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match True
[D] MSR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[... cut ...]
[D] HWP for cpu5 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu6 - write "balance_performance" - read "balance_performance" - match True
[D] HWP for cpu7 - write "balance_performance" - read "balance_performance" - match True
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match True
[D] MSR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0xcc816800dc8168 - read 0xcc816800dc8168 - match True
```

erpalma, 2018-08-22:
> I log out my user and login again.

Hmm I guess this is a OS problem then.

MartinX3, 2018-08-22:
Thank you for your response!

May I ask you, where I can finde the uninstall script?

And how do I know if the script set the vcore correctly?

erpalma, 2018-08-22:
```
[D] Undervolt plane CORE - write 0xecc00000 - read 0xecc00000 - match True
[D] Undervolt plane GPU - write 0x0 - read 0x0 - match True
[D] Undervolt plane CACHE - write 0xecc00000 - read 0xecc00000 - match True
[D] Undervolt plane UNCORE - write 0x0 - read 0x0 - match True
[D] Undervolt plane ANALOGIO - write 0x0 - read 0x0 - match True
```
So it is correct.

To uninstall just issue `systemctl stop lenovo_fix; systemctl disable lenovo_fix; rm -r /opt/lenovo_fix`.

MartinX3, 2018-08-23:
The exact value of -0,150V is set correctly?

So at least I can use your nice tool to adjust my vcore value to cool down my cpu.

Could you write the uninstall manual into the readme? :)

closed 2018-08-30 by DEvil0000