erpalma/throttled #39: Lenovo T470: seems to work?

andyboeh, 2018-08-25:
I've got a T470 with an i7-7600U CPU. Right after bootup, it goes frequently into throttling so I decided to give this fix a try.

What I observed using s-tui is that it can stay way longer in Turbo Boost with the fix applied. The temperature raises to 96C and Power to about 23W (it stays at 19W without the fix).

I'm currently experimenting whether the EC resets the values periodically as well...

DEvil0000, 2018-08-30:
reopen if you have a update about the EC

closed 2018-08-30 by DEvil0000

andyboeh, 2018-09-02:
So, I've got an update on the EC: It resets back to default, at least the thermal limit, that's what I checked using `rdmsr -f 29:24 -d 0x1a2` (jumps back to 25).

To sum up: the script works perfectly on the T470 and makes the frequent "CPU0: Package temperature above threshold, cpu clock throttled" disappear.

What I **do** receive, however, is (seems to be known issue): 
```
thinkpad_acpi: unknown possible thermal alarm or keyboard event received
thinkpad_acpi: unhandled HKEY event 0x6032
thinkpad_acpi: please report the conditions when this event happened to ibm-acpi-devel@lists.sourceforge.net
```