erpalma/throttled #43: X1C6 hangs with lenovo-throttling-fix

krionbsd, 2018-08-29:
I'm using Linux Mint 19 on one of my laptop (X1C6) and after going through all steps and uninstalling thermald, laptop hangs spontaneous during first 2-3 mins after starting, deinstalling throttling fix helps to fix it.

Is it a known issue with Mint?

erpalma, 2018-08-29:
What do you mean with hang exactly?

krionbsd, 2018-08-29:
It's unresponsive completely, I only can reboot it

krionbsd, 2018-08-29:
I forgot to mention, I'm using latest BIOS 1.27 and disabled "Secure Boot" in it.

erpalma, 2018-08-29:
Uh so it's freezed. Did you enabled undervolting?

krionbsd, 2018-08-29:
Yes, I've enabled these settings:

[UNDERVOLT]
# CPU core voltage offset (mV)
CORE: -110
# Integrated GPU voltage offset (mV)
GPU: -90
# CPU cache voltage offset (mV)
CACHE: -110
# System Agent voltage offset (mV)
UNCORE: -90
# Analog I/O voltage offset (mV)
ANALOGIO: 0

MartinX3, 2018-08-29:
And without undervolting you still get the freezing?

krionbsd, 2018-08-29:
Hmm, that's interesting, it runs fine after removing undervolting settings completely, so any hints about it?

DEvil0000, 2018-08-29:
your undervolting is definitivly too high. lower the values by about 20.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/erpalma/lenovo-throttling-fix/issues/43#issuecomment-417038919",
"url": "https://github.com/erpalma/lenovo-throttling-fix/issues/43#issuecomment-417038919",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [erpalma/lenovo-throttling-fix] X1C6 hangs with lenovo-throttling-fix (#43)",
"sections": [
{
"text": "",
"activityTitle": "**Martin Dünkelmann**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@MartinX3",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"erpalma/lenovo-throttling-fix\",\n\"issueId\": 43,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"erpalma/lenovo-throttling-fix\",\n\"issueId\": 43\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/erpalma/lenovo-throttling-fix/issues/43#issuecomment-417038919"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 373742679\n}"
}
],
"themeColor": "26292E"
}
]

krionbsd, 2018-08-29:
ok, thanks for hint, I'll try with -90 and -70 next time.
What surprises me, if I run on battery or AC:

```
sudo rdmsr -f 29:24 -d 0x1a2
20
```

IIRC it should be 15 on battery and 3 on AC.

erpalma, 2018-08-30:
So you are reading 20 (80 'C) while the script is running?

DEvil0000, 2018-08-30:
check with `--debug` if it is getting set correct.
After a config change you need to restart the script.

krionbsd, 2018-08-30:
Yes, I set new values for undervolting like:
```
CORE: -105
GPU: -85
CACHE: -105
UNCORE: -85
ANALOGIO: 0
```
Restarted it:
```
~ systemctl status lenovo_fix.service
● lenovo_fix.service - Stop Intel throttling
   Loaded: loaded (/etc/systemd/system/lenovo_fix.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-08-30 11:37:27 CEST; 3min 45s ago
 Main PID: 5074 (python3)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/lenovo_fix.service
           └─5074 /opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py
```
But I still get value 10 running rdmst on AC:
```
~ sudo rdmsr -f 29:24 -d 0x1a2              
10
```

krionbsd, 2018-08-30:
Thank you @DEvil0000 for hints and clarification, I'll try proposed values

closed 2018-08-30 by krionbsd