erpalma/throttled #45: Compatible with X280

spielkind, 2018-08-30:
Is this compatible/needed with/at the X280?

erpalma, 2018-08-30:
In theory it should be compatible with almost any laptop with Skylake (gen6) or newer CPUs. Is it needed? You can stress test your laptop using `stress`, `stress-ng`, `Prime95` or any other tool and monitor the frequency/temperature using [s-tui ](https://github.com/amanusk/s-tui). Since the critical temperature of our CPU is 100 'C you should expect/want that your CPU starts throttling somewhere above 90 'C. 

As a reference I have set 95 'C on my system in order to get the highest performance possible while on AC.

closed 2018-12-04 by erpalma

fthiery, 2019-01-10:
> > i got confused by the fact that --monitor was also applying the fix, i was expecting it would only report what was the limiting factor
> Yeah, I guess I should check if the tool is already running or not thanks!

Not only that, maybe you should add a --skip-fix switch (or similar) to allow for the monitoring mode to act as "read-only" monitor (if that is even possible !)

> A friend of mine has the same notebook, let me do some tests.

Cool, thanks. What about the "This script forces the CPU package power limit (PL1/2) to 44 W (29 W on battery)" part ?