erpalma/throttled #46: should we rename the tool

DEvil0000, 2018-08-30:
The tool is more targeting general cpu settings and is not related to lenovo in any other way beside their BIOS defaults are bad.
I think we should think about a new name for it which reflects this better.

DEvil0000, 2018-08-30:
beside that how about closing some old solved issues here? would you give me privileges?

erpalma, 2018-08-30:
I totally agree with you. Do you have any ideas?

I've never changed the name of a repo, do you know if there exists some sort of redirection to the new url?

MartinX3, 2018-08-30:
mobile-x86-throttling-fix
hihi

DEvil0000, 2018-08-30:
after a quick search i found those doing more or less the same:
there is a guide [linux-intel-undervolt](https://github.com/mihic/linux-intel-undervolt)
a undervolt tool [undervolt](https://github.com/georgewhewell/undervolt)
a c version of doing the same [intel-undervolt](https://github.com/kitsunyan/intel-undervolt)

They have better names but less users as it seems.
I think the name may contain one of the following words:
intel or cpu
throttle fix
performance
tune or tuning
tweak
tool

I did also never change the name of a repo maybe creating a new one and pushing current master to it, leaving a readme note in the old one may be a workaround if there is no url redirect after rename.

erpalma, 2018-08-30:
There redirection should occur automatically according to [this](https://blog.github.com/2013-05-16-repository-redirects-are-here/).

BTW @DEvil0000 I've added you as a collaborator.

erpalma, 2018-08-31:
Maybe we can copy the Windows utility ThrottleStop with something like LinuxThrottleStop ;)

erpalma, 2018-09-28:
I guess that we should really find a new name :)

intelfx, 2018-11-15:
+1 for this. Just `throttlestop` feels appropriate and consistent enough. Or even throttle-tool, or `throttled` for the daemon itself — after all, you can use it to restrict limits even more ;)

deisi, 2018-12-15:
some brainstorming (no research done):

- dethrottle
- antithrottle
- nothrottle

- intelaccelerator

or maybe some strange unrelated acronym
- sits (sane intel temperature settings)

closed 2019-01-18 by erpalma