erpalma/throttled #48: file path problems

bluePain, 2018-08-30:
Working on Asus UX430UN(R).

Os info: `4.18.5-arch1-1-ARCH #1 SMP PREEMPT Fri Aug 24 12:48:58 UTC 2018 x86_64 GNU/Linux`

1. `SYSFS_POWER_PATH` should be parametric and should be read from the config file. In my laptop this path is `/sys/class/power_supply/AC0/online`.
2. `virtualenv` command in install.sh script is not necessary. I don't have it installed on my laptop and calling pip directly with python such as `python3 -m pip install -r requirements.txt`
3. `/opt/lenovo_fix/venv/bin/python3` is a bit odd in service file. Can be replaced with `/usr/bin/python3` without hesitation.

erpalma, 2018-08-30:
1. Uh strange! I can move it to the config file yes. I can also check the existence of `AC[0-9]*`.
1. Sorry but I do not agree at all. Installing python modules globally with pip is not the best solution and can also conflict with the modules installed by the package manager of your distro. Per-project dependencies should be handled using a virtualenv. 
1. I hard-coded the path to avoid setting the `VIRTUAL_ENV` var in the service file and using `/usr/bin/env python3`.

bluePain, 2018-08-31:
1. As Devil stated it may depend on the kernel and may change in the future again. I believe trying to find the correct path is a burden on the software. You can try to check the predefined paths and if you cannot find any of them then you can try to read from the config file.
2. You're right. `virtualenv` way better approach than calling `python3 -m pip` directly to keep the os clean.
3. Since this is related with the 2. item we can ignore this.

Additionally, you can modify the install script to check if the required commands are installed in the system such as `python-pip` or `virtualenv`.

erpalma, 2018-08-31:
I do agree to add `SYSFS_POWER_PATH` to the config file, but I would like to cover 99% of the usage case with the default implementation. Not every user knows the `sysfs` structure thus I prefer to avoid forcing the user to configure a parameter that I can possibly guess. My idea is to try with `AC` and `AC0` (which is what many other libs are doing) unless the user overrides the value in the config file. We can leave the entry commented out like

`#SYSFS_POWER_PATH = "/sys/class/power_supply/AC0/online"`

> Additionally, you can modify the install script to check if the required commands are installed in the system such as python-pip or virtualenv.

Good idea!

closed 2018-09-14 by erpalma