erpalma/throttled #52: Installation fails on Ubuntu 18.04 amd64

bjoern-tantau, 2018-09-05:
It looks like the install script needs the i386 version of the dependencies. I was able to install most of them by appending `:i386` to the apt-get command from the readme. But Ubuntu won't install `libgirepository1.0-dev:i386` at the moment.

```
sudo apt install libgirepository1.0-dev:i386
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 libgirepository1.0-dev:i386 : Hängt ab von: gobject-introspection:i386 (= 1.56.1-1) soll aber nicht installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
```

This is the output of the install script:

```
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
Building virtualenv...
Already using interpreter /usr/bin/python3
Using base prefix '/usr'
New python executable in /opt/lenovo_fix/venv/bin/python3
Not overwriting existing python script /opt/lenovo_fix/venv/bin/python (you must use /opt/lenovo_fix/venv/bin/python3)
Installing setuptools, pkg_resources, pip, wheel...done.
Requirement already satisfied: configparser==3.5.0 in ./venv/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (3.5.0)
Requirement already satisfied: dbus-python==1.2.8 in ./venv/lib/python3.6/site-packages (from -r requirements.txt (line 2)) (1.2.8)
Requirement already satisfied: psutil==5.4.7 in ./venv/lib/python3.6/site-packages (from -r requirements.txt (line 3)) (5.4.7)
Collecting PyGObject==3.30.0 (from -r requirements.txt (line 4))
  Using cached https://files.pythonhosted.org/packages/40/12/071e113583e826c9c06b06779730f2b5c3aae1d1a2cf715f25daa8f62ec6/PyGObject-3.30.0.tar.gz
Requirement already satisfied: pycairo>=1.11.1 in ./venv/lib/python3.6/site-packages (from PyGObject==3.30.0->-r requirements.txt (line 4)) (1.17.1)
Building wheels for collected packages: PyGObject
  Running setup.py bdist_wheel for PyGObject ... error
  Complete output from command /opt/lenovo_fix/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-hfnpk6p1/PyGObject/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-aca1qxbq --python-tag cp36:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.6
  creating build/lib.linux-x86_64-3.6/pygtkcompat
  copying pygtkcompat/__init__.py -> build/lib.linux-x86_64-3.6/pygtkcompat
  copying pygtkcompat/generictreemodel.py -> build/lib.linux-x86_64-3.6/pygtkcompat
  copying pygtkcompat/pygtkcompat.py -> build/lib.linux-x86_64-3.6/pygtkcompat
  creating build/lib.linux-x86_64-3.6/gi
  copying gi/_propertyhelper.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/__init__.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/types.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_signalhelper.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/importer.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_constants.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_gtktemplate.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_option.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_compat.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_ossighelper.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/pygtkcompat.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/_error.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/docstring.py -> build/lib.linux-x86_64-3.6/gi
  copying gi/module.py -> build/lib.linux-x86_64-3.6/gi
  creating build/lib.linux-x86_64-3.6/gi/repository
  copying gi/repository/__init__.py -> build/lib.linux-x86_64-3.6/gi/repository
  creating build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/__init__.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/Gdk.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/GObject.py -> build/lib.linux-x86_64-3.6/gi/overrides
  copying gi/overrides/keysyms.py -> build/lib.linux-x86_64-3.6/gi/overrides
[... 46 lines cut ...]
    creating build/lib.linux-x86_64-3.6/gi/repository
    copying gi/repository/__init__.py -> build/lib.linux-x86_64-3.6/gi/repository
    creating build/lib.linux-x86_64-3.6/gi/overrides
    copying gi/overrides/__init__.py -> build/lib.linux-x86_64-3.6/gi/overrides
    copying gi/overrides/Gdk.py -> build/lib.linux-x86_64-3.6/gi/overrides
    copying gi/overrides/GObject.py -> build/lib.linux-x86_64-3.6/gi/overrides
    copying gi/overrides/keysyms.py -> build/lib.linux-x86_64-3.6/gi/overrides
    copying gi/overrides/Gtk.py -> build/lib.linux-x86_64-3.6/gi/overrides
    copying gi/overrides/GIMarshallingTests.py -> build/lib.linux-x86_64-3.6/gi/overrides
    copying gi/overrides/GLib.py -> build/lib.linux-x86_64-3.6/gi/overrides
    copying gi/overrides/Pango.py -> build/lib.linux-x86_64-3.6/gi/overrides
    copying gi/overrides/GdkPixbuf.py -> build/lib.linux-x86_64-3.6/gi/overrides
    copying gi/overrides/Gio.py -> build/lib.linux-x86_64-3.6/gi/overrides
    running build_ext
    Package gobject-introspection-1.0 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `gobject-introspection-1.0.pc'
    to the PKG_CONFIG_PATH environment variable
    Package 'gobject-introspection-1.0', required by 'world', not found
    Command '('pkg-config', '--print-errors', '--exists', 'gobject-introspection-1.0 >= 1.46.0')' returned non-zero exit status 1.
    
    Try installing it with: 'sudo apt install libgirepository1.0-dev'
    
    ----------------------------------------
Command "/opt/lenovo_fix/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-hfnpk6p1/PyGObject/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-p64nygsx/install-record.txt --single-version-externally-managed --compile --install-headers /opt/lenovo_fix/venv/include/site/python3.6/PyGObject" failed with error code 1 in /tmp/pip-install-hfnpk6p1/PyGObject/
```

erpalma, 2018-09-05:
> It looks like the install script needs the i386 version of the dependencies.

Can you explain please?

bjoern-tantau, 2018-09-06:
Before it was complaining about not being able to find dependencies, like cairo, dbus or dbus-glib, although I had them installed. These errors went away, when I installed the i386 versions by running.

```
sudo apt install libdbus-1-dev:i386 libdbus-glib-1-dev:i386 libcairo2-dev:i386
```

But installing libgirepository1.0-dev:i386 fails.

erpalma, 2018-09-06:
I do not need any 32-bit dependency. I guess you have either python:i386 or some other package that is running by default in 32-bit mode, thus requiring you to install all the deps in 32-bit too.

bjoern-tantau, 2018-09-06:
Thanks, for some reason I had pkgconf:i386 installed. Installing pkgconf instead solved the issue for me.

commit 6516f446 (2018-09-03): config values can be commented out to disable (fix #52)
closed 2018-09-06 by bjoern-tantau