erpalma/throttled #55: Arch Install - Incorrect vitrualenv location in readme.md & virtualenv not including dbus (module import error)

digitaltommy, 2018-09-30:
> The scripts is now running with Python3 by default (tested w/ 3.6) and a virtualenv is automatically created in /opt/lenovo_fix. Python2 should probably still work.

Installed using:
```
yaourt -S lenovo-throttling-fix-git
```
virtualenv location was /usr/bin/env, not in /opt/

---

Also had module import error (ModuleNotFoundError: No module named 'dbus') for dbus, python-dbus was installed.

Changed the script to use my python install instead of the virtualenv and had no issues.

From:
`
#!/usr/bin/env python
`
To:
`
#!/usr/bin/python3
`

erpalma, 2018-09-30:
Actually I'm not the maintainer of the arch package. Also `/usr/bin/env` is not the location of the virtualenv (which is not created by the Arch package BTW) but the `env` helper program. I guess that `#!/usr/bin/env python` is not working for you because you have set 2.7 as the system default python version. Right?

digitaltommy, 2018-09-30:
My mistakes, apologies. I though I'd put the issues I had here in case it helped anyone else.

> Also /usr/bin/env is not the location of the virtualenv (which is not created by the Arch package BTW) but the env helper program.

Ah this makes way more sense, apologies my beginner assumptions!

> I guess that #!/usr/bin/env python is not working for you because you have set 2.7 as the system default python version. Right?

When I run /usr/bin/env python or python I get Python 3.7.0 as default.

Thanks for your help.