erpalma/throttled #58: Any experiences with E480/580?

uli42, 2018-10-12:
SSIA

Leryan, 2019-12-27:
Hello lately there,

Yes, on my E480 (`LENOVO_MT_20KN_BU_Think_FM_ThinkPad E480` as `dmidecode` shows) it works perfectly.

I'm able to modify the max temp before throttling and it really does a good job.

I'm finally able to run GoLand without excessive lags :p

Ping @erpalma this device may be added to the supported list. If you want more tests i'm here to help :)

closed 2019-12-30 by erpalma

uli42, 2019-12-30:
As E480 and E580 are pretty identical (except for the display size) I think it would work there, too. So I suggest to add E580 to the list as well.

uli42, 2020-01-02:
Thanks for adding E580 to the list. Today I finally found time to test this on such a machine (E580 - 20KSCTO1WW -  LENOVO_MT_20KS_BU_Think_FM_ThinkPad). Unfortunately it does not really make a difference here. Running s-tui stress test leads to hitting the 95 degree border and the frequency is about 2800 then (the system has an i7-8550U). I think this is related to a BIOS fix Lenovo made. I do not have any more details, but the changelog reports "- (New) Improved the performance when the system were under high loading" for BIOS 1.33 (I have 1.35) so this might be an non-issue now. Also `rdmsr -f 29:24 -d 0x1a2` showed `3` even before installing the throttled.

I think it could help if you added a short instruction that explains how to test if one is affected by the issue(s) your tool tries to fix.

Leryan, 2020-01-02:
I have the very latest BIOS from lenovo on my E480 and it does make a difference.

However the machine actually reaches 95 degrees on full load, making the cpus throttle slowly from 3400 to 2800 like you.

One way to avoid this is to raise the max temp in lenovo_fix.conf to 98 for example, it may be enough.

commit e9ec389f (2019-12-30): Add Thinkpad E480 to support list (fix #58)