erpalma/throttled #60: i want run it 

chenjiangui, 2018-10-25:


Xplosm, 2018-10-25:
Well, since this tool was created an issue with the Lenovo series T480* in mind I don't see any benefit for either OSX not Apple architecture.

chenjiangui, 2018-10-26:
i install Hackintosh in my T480,  but the max tdp is 15w, only turbo frequency about 15S, second the tdp will be 15w and frequency be 2.4GHz, in the windows, xtu can change tdp to 45W, and always turbo frequency, so i want change the tdp on os-x

DEvil0000, 2018-10-26:
i guess you need to make it work then. waiting for a pull request.

erpalma, 2018-10-27:
On OSX it's a bit difficult to get this script working. You need a custom kernel module for the WRMSR instruction. `/dev/mem` should be accessible via `sudo nvram boot-args="kmem=1"`. Then, porting the `/sys`, dbus, systemd stuff should be easier.

nariox, 2018-11-26:
Hi @chenjiangui,

This project hasn't seem updates in a year, but should be a good starting point.