erpalma/throttled #61: Question about PL2_Tdp_W and BATTERY values in general

hyphone, 2018-10-30:
Hello and good day,

this is not an issue but a question about the BATTERY value of PL2_Tdp_W.

On default it's also set to 44. Shouldn't this also be set to 29 like PL1_Tdp_W?

And another question in general:
When I have an i7-8550U the normal TLP is 15 watts right? So it's still lower than the default battery value of 29 here?
Is it possible to set the BATTERY values of PL1_Tdp_W and PL2_Tdp_W to 15 to even save more juice?

Thank you in advance and all the best

DEvil0000, 2018-10-30:
Those values are only applied for a short time turbo boost. You could limit it more but I am not sure how much sense this makes.

erpalma, 2018-10-30:
> Is it possible to set the BATTERY values of PL1_Tdp_W and PL2_Tdp_W to 15 to even save more juice?

Well you can do it of course, but the best way to lower the power consumption is to use [TLP](https://github.com/linrunner/TLP). 

> On default it's also set to 44. Shouldn't this also be set to 29 like PL1_Tdp_W?

I set 29/44 W for battery to ensure that if you use TLP and temporary need more power (`tlp ac` command), then this script is not a power limiting factor. The `PL2_Tdp_W` value is, by default, valid for a 2ms time window, so I don't think it would make sense to lower it. It is used only for very short computational power spikes.

hyphone, 2018-10-30:
Thank you both for the detailed answers!

closed 2018-11-06 by erpalma