erpalma/throttled #62: Note about C-states on XPS 9370.

c-edw, 2018-10-31:
This isn't a bug, but might be useful information to have for anyone passing by.
On my XPS 9370, I've been seeing a lot of kernel messages complaining about thermal throttling (package/core above temperature), despite it being only 40-50 degrees. One of the widgets in my window manager was reporting the CPU temperature as 90 degrees even though it wasn't.
For some reason, disabling C-states in the UEFI fixes this. I'm inclined to say it's a firmware bug.

hyphone, 2018-10-31:
This explains something. Are you on BIOS 1.5.1?
Disabling C-States will disable all the CPU powersaving features so frequency scaling won't work anymore. Is that correct?

c-edw, 2018-10-31:
I currently have 1.4.0, and 1.5.1 is just a few IME updates, so an update is unlikely to fix anything. According to the information in the UEFI, C-states are for additional power savings during system idle or hibernation. P-states are still working and it will underclock the CPU to save power. Functionally, I've seen no difference from disabling C-states, but it's resolved all of the problems I was having.

hyphone, 2018-10-31:
Thanks for sharing. I'll have an eye on this from now on if this also happens on my machine.

hyphone, 2018-12-03:
This also doesn't happen anymore when a someone does a cold boot (with C-States enabled).

c-edw, 2019-01-01:
Closing this. I think it happens with SpeedStep still for me, but not C-states. Weird stuff. I think the kernel messages can be ignored.

closed 2019-01-01 by c-edw