erpalma/throttled #66: Error in check_kernel

sanderboom, 2018-11-07:
Hello!

Eager to test the [throttling monitoring](https://github.com/erpalma/lenovo-throttling-fix/commit/9636a5de8327fc2d6874b97da93e6169d9d78588) (awesome!) I reinstalled the tool completely (default config). 

Upon starting the script I run into the following error message: 

```shell
sander@t480s ~
λ sudo /usr/lib/lenovo-throttling-fix/lenovo_fix.py
Traceback (most recent call last):
  File "/usr/lib/lenovo-throttling-fix/lenovo_fix.py", line 478, in check_kernel
    with open(os.path.join('/boot', 'config-{:s}'.format(uname()[2]))) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/boot/config-4.19.0-4-MANJARO'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/lenovo-throttling-fix/lenovo_fix.py", line 591, in <module>
    main()
  File "/usr/lib/lenovo-throttling-fix/lenovo_fix.py", line 510, in main
    check_kernel()
  File "/usr/lib/lenovo-throttling-fix/lenovo_fix.py", line 483, in check_kernel
    kernel_config = f.read()
  File "/usr/lib/python3.7/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8b in position 1: invalid start byte
```

I'm not a Python dev, but are you sure you can do this:

```python
            with open(os.path.join('/proc', 'config.gz')) as f:
                kernel_config = f.read()
```

on a binary file?
```shell
λ file /proc/config.gz
/proc/config.gz: gzip compressed data, max compression, from Unix, original size 221842
```
On a side note I was fooled for a moment by `vim` just opening `/proc/config.gz` in extracted/text format.

The `CONFIG_X86_MSR=` & `CONFIG_DEVMEM=y` are present.

Thanks for looking into this!

aussetg, 2018-11-07:
Still doesn't work:

```
nov. 07 13:38:28 laptop lenovo_fix.py[11540]:     main()
nov. 07 13:38:28 laptop lenovo_fix.py[11540]:   File "/usr/lib/lenovo-throttling-fix/lenovo_fix.py", line 544, in main
nov. 07 13:38:28 laptop lenovo_fix.py[11540]:     check_kernel()
nov. 07 13:38:28 laptop lenovo_fix.py[11540]:   File "/usr/lib/lenovo-throttling-fix/lenovo_fix.py", line 490, in check_kernel
nov. 07 13:38:28 laptop lenovo_fix.py[11540]:     elif not re.search('CONFIG_DEVMEM=y', kernel_config):
nov. 07 13:38:28 laptop lenovo_fix.py[11540]:   File "/usr/lib/python3.7/re.py", line 183, in search
nov. 07 13:38:28 laptop lenovo_fix.py[11540]:     return _compile(pattern, flags).search(string)
nov. 07 13:38:28 laptop lenovo_fix.py[11540]: TypeError: cannot use a string pattern on a bytes-like object
```

erpalma, 2018-11-07:
Meh, I still hate py3... gzip.open defaults to 'rb' mode. 

I know that `/proc/config.gz` requires the `configs` module, I can try to load it but the point is that this is already a fallback. Don't you have `/boot/config-xxx`?

sanderboom, 2018-11-07:
> Don't you have /boot/config-xxx

Nope. Relative fresh/default install of Manjaro KDE here. 

```shell
sander@t480s /boot
λ find -name "config-"
sander@t480s /boot
λ uname -a
Linux t480s 4.19.0-4-MANJARO #1 SMP PREEMPT Thu Nov 1 15:20:23 UTC 2018 x86_64 GNU/Linux
```

> Meh, I still hate py3... gzip.open defaults to 'rb' mode.

Maybe use a gunzipped tempfile as a last fallback?

sanderboom, 2018-11-07:
Yea, tried that before as well, nothing here :/

```shell
sander@t480s /boot
λ find -name "config*"
./grub/x86_64-efi/configfile.mod
sander@t480s /boot
λ ls
efi         grub             vmlinuz-4.14-x86_64   vmlinuz-4.19-x86_64   vmlinuz-4.18-x86_64   initramfs-4.14-x86_64.img           initramfs-4.18-x86_64.img           initramfs-4.19-x86_64.img
memtest86+  intel-ucode.img  linux414-x86_64.kver  linux419-x86_64.kver  linux418-x86_64.kver  initramfs-4.14-x86_64-fallback.img  initramfs-4.18-x86_64-fallback.img  initramfs-4.19-x86_64-fallback.img
```

erpalma, 2018-11-07:
Hmm, nice to know thanks. The `/proc/config.gz` way should be working now.

sanderboom, 2018-11-07:
Awesome! Thanks for the quick fix. Man, you do great stuff on your bad days ;-)

This **is** sexy, or what?!

[image]

erpalma, 2018-11-07:
Ahahah thanks a lot, very appreciated! :D

I'll add some more monitoring stuff and polish the output as soon as I can, stay tuned ;)