erpalma/throttled #72: X1 Yoga Support

ElijahLynn, 2018-11-22:
Having this issue on X1 Yoga (3rd generation), installed your package in Arch and still seeing the log with the messages. Can we get the X1 Yoga supported?

ejgallego, 2018-11-22:
I have a X1 3rd gen and it is working beautifully , have you disabled thermals?

erpalma, 2018-11-22:
Are you seeing thermal events like this?

`CPU0: Package temperature above threshold, cpu clock throttled (total events = 103832)`

That's "normal". The EC/BIOS is still reporting those events when CPU temp goes above the threshold you set in the config file.

ElijahLynn, 2018-11-22:
@erpalma okay, that is good to know, thanks. 

@ejgallego Thanks for that confirmation, I see `[acpi_thermal_pm]` running but not `thermald`.

ejgallego, 2018-11-22:
The way to check things are working fine is with s-tui for example. These new CPUs are a bit strange in that performance is pretty variable, but I get 2.7 GHz with all cores stable, without underclocking

DEvil0000, 2018-11-22:
> @ejgallego Thanks for that confirmation, I see [acpi_thermal_pm] running but not thermald.

AFAIK  [acpi_thermal_pm] is the kernel process doing all the acpi thermal power management stuff. I don't think its changeing things on its own but I think its providing the interface (proc) for other processes to change things there. So maybe you have some process changeing things. Maybe its a good idea to stop everything beside the throttle stop and the s-tui until it works as expected.

erpalma, 2018-11-22:
Try `--monitor` mode and s-tui under stress!

hyphone, 2018-12-30:
For me it works as intended with this machine. I haven't encountered any issues, yet.

@ElijahLynn  The CPU hits the critical temperatures quite quickly but that's the cause of the insufficient cooling system (just one heat pipe and one fan compared to two heat pipes and two fans in the XPS 9370).

ElijahLynn, 2019-08-20:
Nice, I somehow must have had it installed incorrectly. I just tried seeing if I had it enabled and eventually saw there was an AUR package by @felixonmars. And followed those instructions and instantly a video export I was doing jumped up in CPU/Temp and Power consumption!

Also when I installed the AUR of this package it said it was going to conflict with the git version of this package I had installed which I replaced. 

[image]

Big thanks everyone for this!

ElijahLynn, 2019-08-20:
> Nice, I somehow must have had it installed incorrectly. I just tried seeing if I had it enabled and eventually saw there was an AUR package by @felixonmars. And followed those instructions and [...]

I suppose it isn't really getting to 4.2 Ghz but not really sure of the requirements to achieve that are. Should I actually see it hit 4.2 Ghz in `s-tui`?

closed 2019-08-20 by ElijahLynn

erpalma, 2019-08-20:
4.2 GHz can only be reached in single core usage. But I guess that your main problem is cooling now. As you can see you are reaching 95 deg at just 2.2 GHz, so you are now thermal limited. Try re-pasting your CPU, but I think you will only get 100/200 MHz more.

ElijahLynn, 2019-08-20:
> 4.2 GHz can only be reached in single core usage. But I guess that your main problem is cooling now. As you can see you are reaching 95 deg at just 2.2 GHz, so you are now thermal limited. Try re-pasting your CPU, but I think you will only get 100/200 MHz more.

Thanks for that, very helpful to know 4.2 Ghz is a single core feature. 

Also great to know that 95 deg is the upper limit before thermal limiting kicks in. 

I'll try running a fan on the Yoga and also open the lid as it is currently docked. Maybe I can squeeze more out that way!

ejgallego, 2019-08-20:
As a reference, my X1 Yoga 3rd generation, with standard cooling and voltage does get to 4.2 in single core, and will all cores is around 2.5-3Ghz depending on the ambient temp.

erpalma, 2019-08-20:
According to what @ejgallego wrote I would definitely try re-pasting the CPU then!

ejgallego, 2019-08-20:
Note that my temp threshold is at 100º , in the s-tui display there it seems it peaks at 95?

ElijahLynn, 2019-08-20:
> According to what @ejgallego wrote I would definitely try re-pasting the CPU then!

Don't think I can void the warranty like that just yet, as it is also owned by my employer.

ElijahLynn, 2019-08-20:
> As a reference, my X1 Yoga 3rd generation, with standard cooling and voltage does get to 4.2 in single core, and will all cores is around 2.5-3Ghz depending on the ambient temp.

Super helpful, thank you! How did you measure the 4.2 on single core? I assuming not s-tui.

ejgallego, 2019-08-21:
I was indeed using s-tui, note that it is never 4.2 exact , it goes up and down, but close to 4.1 Ghz most of the time.

With recommended undervolting my system is stable at 3.75 Ghz now (1 core) and 2.9 Ghz with all cores [it can take a bit more for the start, but indeed when the laptop gets hot it settles down on that]

That's on my room which is quite warm now [summer], but indeed thermal is the limit of this laptop, I didn't re-paste the CPU but for sure that should help. Just going to the terrace with a bit more of fresh air did got me almost 0.1 extra Ghz.

ejgallego, 2019-08-21:
Couple of pics in case they help:
[image]
[image]

erpalma, 2019-08-21:
> Note that my temp threshold is at 100º , in the s-tui display there it seems it peaks at 95?

It peaks at 95 because that's the trip temperature set in throttled. Once you reach 100 deg your CPU shuts down ;)