erpalma/throttled #73: P1 Throttling in battery mode

phycent, 2018-11-22:
It looks like the laptop is thermal throttling at 60 Celsius. 
rdmsr -f 29:24 -d 0x1a2 gives 5, which is the default value on lenovo-throttling-fix

phycent, 2018-11-22:
I see the power never exceeds 10W on battery and occasionally it says Current is Limited. 
Thermal: OK - Power: OK - Current: OK - Cross-comain (e.g. GPU): OK  ||  VCore: 713 mV - Package: 9.8 W - Graphics: 0.1 W - DRAM: 1.2 W            
Thermal: OK - Power: OK - Current: OK - Cross-comain (e.g. GPU): OK  ||  VCore: 713 mV - Package: 8.9 W - Graphics: 0.1 W - DRAM: 1.2 W           
Thermal: OK - Power: OK - Current: OK - Cross-comain (e.g. GPU): OK  ||  VCore: 714 mV - Package: 9.4 W - Graphics: 0.1 W - DRAM: 1.2 W          
Thermal: OK - Power: OK - Current: LIM - Cross-comain (e.g. GPU): OK  ||  VCore: 989 mV - Package: 9.1 W - Graphics: 0.1 W - DRAM: 1.1 W          
Thermal: OK - Power: OK - Current: LIM - Cross-comain (e.g. GPU): OK  ||  VCore: 998 mV - Package: 9.1 W - Graphics: 0.1 W - DRAM: 1.2 W

DEvil0000, 2018-11-22:
I don't know why your current limit gets hit - there might be a stupid reason like: bad or weak power supply or bad regulators on the mainboard but it might also be some other process limiting you.

erpalma, 2018-11-22:
I agree, this is very strange at just 9W! What about on Windows?

phycent, 2018-11-22:
This only happens in battery mode, not in AC mode.
On windows, it is showing the same behavior: https://imgur.com/a/KTii2Ms

phycent, 2018-11-22:
On Windows, after installing Lenovo Power Management, the system can output ~30W of power.I think the same thing can be done in Linux too.

> This only happens in battery mode, not in AC mode.
> On windows, it is showing the same behavior: https://imgur.com/a/KTii2Ms

DEvil0000, 2018-11-23:
very strange but it could be reasonable in some conditions:
low or high battery or ambient temperatures, small battery, degraded battery, some strange power saving forced from somewhere

erpalma, 2018-11-23:
In BIOS you can set the power profile on battery mode. Try to set it to the same value of AC. I guess Lenovo Power Management is fiddling some values in the EC.

phycent, 2018-11-23:
I already matched the power policy on battery the same as on AC before installing this service.

Now I got it running at a maximum of 20 W. Not entirely sure about what I did. But I installed dbus-python and install the whole service again.

closed 2018-11-23 by phycent