erpalma/throttled #77: Limited Power on XPS 9370

hyphone, 2018-12-03:
Hello and good day,

I'm testing a little more at the moment with the XPS 9370 and the new --monitor feature.

It seems that Power is limited most of the time with the charger plugged in. I'm using the official 45W charger and an XPS 15 charger with a lot more juice and an USB-C adapter. 

s-tui says it just consumes 11W of 18W max.

On battery the Power value is OK because of the lower TPD (I guess).

Perhaps someone has a clue where I could look further.

[image]

erpalma, 2018-12-03:
Can you please check with `--debug` if all the values are correctly written?

hyphone, 2018-12-03:
Thank you for the quick response.

It seems so - this is my output:

[D] core 0 thermal status: thermal limit status = 0
[D] core 0 thermal status: thermal limit log = 0
[D] core 0 thermal status: prochot or forcepr status = 0
[D] core 0 thermal status: prochot or forcepr log = 0
[D] core 0 thermal status: crit temp status = 0
[D] core 0 thermal status: crit temp log = 0
[D] core 0 thermal status: thermal threshold1 status = 0
[D] core 0 thermal status: thermal threshold1 log = 0
[D] core 0 thermal status: thermal threshold2 status = 0
[D] core 0 thermal status: thermal threshold2 log = 0
[D] core 0 thermal status: power limit status = 0
[D] core 0 thermal status: power limit log = 0
[D] core 0 thermal status: current limit status = 0
[D] core 0 thermal status: current limit log = 0
[D] core 0 thermal status: cross domain limit status = 0
[D] core 0 thermal status: cross domain limit log = 0
[D] core 0 thermal status: cpu temp = 59
[D] core 0 thermal status: temp resolution = 1
[D] core 0 thermal status: reading valid = 1
[D] core 1 thermal status: thermal limit status = 0
[D] core 1 thermal status: thermal limit log = 0
[D] core 1 thermal status: prochot or forcepr status = 0
[D] core 1 thermal status: prochot or forcepr log = 0
[D] core 1 thermal status: crit temp status = 0
[D] core 1 thermal status: crit temp log = 0
[D] core 1 thermal status: thermal threshold1 status = 0
[D] core 1 thermal status: thermal threshold1 log = 0
[D] core 1 thermal status: thermal threshold2 status = 0
[D] core 1 thermal status: thermal threshold2 log = 0
[D] core 1 thermal status: power limit status = 0
[D] core 1 thermal status: power limit log = 0
[D] core 1 thermal status: current limit status = 0
[D] core 1 thermal status: current limit log = 0
[D] core 1 thermal status: cross domain limit status = 0
[D] core 1 thermal status: cross domain limit log = 0
[D] core 1 thermal status: cpu temp = 57
[D] core 1 thermal status: temp resolution = 1
[D] core 1 thermal status: reading valid = 1
[D] core 2 thermal status: thermal limit status = 0
[D] core 2 thermal status: thermal limit log = 0
[D] core 2 thermal status: prochot or forcepr status = 0
[D] core 2 thermal status: prochot or forcepr log = 0
[D] core 2 thermal status: crit temp status = 0
[D] core 2 thermal status: crit temp log = 0
[D] core 2 thermal status: thermal threshold1 status = 0
[D] core 2 thermal status: thermal threshold1 log = 0
[D] core 2 thermal status: thermal threshold2 status = 0
[D] core 2 thermal status: thermal threshold2 log = 0
[D] core 2 thermal status: power limit status = 0
[D] core 2 thermal status: power limit log = 0
[D] core 2 thermal status: current limit status = 0
[D] core 2 thermal status: current limit log = 0
[D] core 2 thermal status: cross domain limit status = 0
[D] core 2 thermal status: cross domain limit log = 0
[D] core 2 thermal status: cpu temp = 59
[D] core 2 thermal status: temp resolution = 1
[D] core 2 thermal status: reading valid = 1
[D] core 3 thermal status: thermal limit status = 0
[D] core 3 thermal status: thermal limit log = 0
[D] core 3 thermal status: prochot or forcepr status = 0
[D] core 3 thermal status: prochot or forcepr log = 0
[D] core 3 thermal status: crit temp status = 0
[D] core 3 thermal status: crit temp log = 0
[D] core 3 thermal status: thermal threshold1 status = 0
[D] core 3 thermal status: thermal threshold1 log = 0
[D] core 3 thermal status: thermal threshold2 status = 0
[... 65 lines cut ...]
[D] core 6 thermal status: temp resolution = 1
[D] core 6 thermal status: reading valid = 1
[D] core 7 thermal status: thermal limit status = 0
[D] core 7 thermal status: thermal limit log = 0
[D] core 7 thermal status: prochot or forcepr status = 0
[D] core 7 thermal status: prochot or forcepr log = 0
[D] core 7 thermal status: crit temp status = 0
[D] core 7 thermal status: crit temp log = 0
[D] core 7 thermal status: thermal threshold1 status = 0
[D] core 7 thermal status: thermal threshold1 log = 0
[D] core 7 thermal status: thermal threshold2 status = 0
[D] core 7 thermal status: thermal threshold2 log = 0
[D] core 7 thermal status: power limit status = 0
[D] core 7 thermal status: power limit log = 0
[D] core 7 thermal status: current limit status = 0
[D] core 7 thermal status: current limit log = 0
[D] core 7 thermal status: cross domain limit status = 0
[D] core 7 thermal status: cross domain limit log = 0
[D] core 7 thermal status: cpu temp = 59
[D] core 7 thermal status: temp resolution = 1
[D] core 7 thermal status: reading valid = 1
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match OK
[D] CONFIG_TDP_CONTROL - write 0x0 - read 0x0 - match OK
[D] MSR PACKAGE_POWER_LIMIT - write 0xcc816000dc8160 - read 0xcc816000dc8160 - match OK
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0xcc816000dc8160 - read 0xcc816000dc8160 - match OK

galfwender, 2018-12-08:
I see the same issue on my 9370 after upgrading to recent version (I had not done so for several months).
If I boot with AC power connected, it limits power as per the configuration file.  I have to disconnect and reconnect AC power for it to recognise it is there.
Despite this, "cat /sys/class/power_supply/AC*/online" returns the correct value at all times.

hyphone, 2018-12-08:
reconnecting the power cord does not have an effect for me.

When it's connected it's limited.
When I disconnect the AC the settings of the battery section are working fine.
When I connect the AC it's limited again.

Thunderbolt ports on the left and the USB-C on the right don't make a difference.

hyphone, 2018-12-08:
I just flashed the older BIOS 1.5.1 again (had 1.6.3 before) and the issue persists.

erpalma, 2018-12-09:
> I see the same issue on my 9370 after upgrading to recent version (I had not done so for several months).
> If I boot with AC power connected, it limits power as per the battery section of the configuration file. I have to disconnect and reconnect AC power for it to detect it and remove the power limit.
> Despite this, "cat /sys/class/power_supply/AC*/online" returns the correct value at all times.
> I am not sure if it makes any difference, but power is connected via a Thunderbolt / USB-C hub.

This sounds like a bug to me. It might be related to the event-based code which previously was just constantly polling the value from sysfs.

hyphone, 2018-12-09:
@erpalma Thanks for the hint.
I'm using v0.3 for now and it works fine.

erpalma, 2018-12-11:
I've just updated the monitor mode to print the power source. This way you can check if there is a mismatch and thus a bug in that code.

erpalma, 2018-12-30:
I also observed a similar behavior a couple of times, but it was related to TLP. Restarting the daemon fixed the issue.

hyphone, 2018-12-30:
I for myself hadn't TLP installed. I used powertop --autotune on boot.

galfwender, 2018-12-30:
I repeated test and rather than disconnect/reconnect AC, tried starting/stopping TLP service as well as manually running "tlp true" and "tlp false" but this did not do anything, the problem still existed.

I then disabled TLP and repeated the test by disconnecting/reconnecting AC and got the same result as yesterday:

- Booting with AC connected seems to enforce 18W / 11W power limits that are not in config file.  Disconnecting and reconnecting AC enforces correct specified limits. (debug log attached)
- Booting with AC disconnected applies power limit of 15W correctly as specified in config file.  When AC is later connected it correctly applies the higher limit specified in config file.

If you look closely you can see when it kicks in to life, momentarily applying the 15W limit whilst on battery immediately before switching to AC power.

[lenovo2.log](https://github.com/erpalma/lenovo-throttling-fix/files/2717384/lenovo2.log)
[lenovo_fix.conf.txt](https://github.com/erpalma/lenovo-throttling-fix/files/2717390/lenovo_fix.conf.txt)
[image]