erpalma/throttled #78: Scaling sometimes dropping to base frequency

tbart, 2018-12-04:
I've got a T470s here (which can be added to the list of compatible devices) which basically works pretty well without this tool but still only scales to 3.4GHz of 3.9GHz in stock configuration (thermal limit only, capped to 75°C).
When I just undervolt (-100mV) without this tool (just a one shot wrmsr) I can get to nearly 3.7GHz (below 75°C), which is pretty nice.

I can squeeze out another 100MHz or so if capping to 90°C (don't want to go any further), but I have to raise vcore at the same time (and P=U^2/R), as compiling a kernel results in segfaults, so this does not really buy me a lot apart from increased power consumption and thermal paste dryout.

What additionally happens if I use this tool is that sometimes all CPUs fall to the base frequency (~800MHz) and do not leave it anymore (that's also mentioned in the reddit thread I think) until I stop compilation and start it again. This never happens when just undervolting. I *think* I can make it happen a lot earlier if I decrease the Update_Rate_s. It seems like some kind of race condition between the EC and this tool, but I might be wrong.
I don't know if this can be solved, most likely #70 will be the correct way to solve this once and for all.

erpalma, 2018-12-04:
Honestly this never happened to my T480s so it's kinda difficult to debug. It is a bit strange that this is "solved" just by re-starting the compilation. Did you tried with `HWP_Mode: True`?

DEvil0000, 2018-12-04:
Sounds a bit like the issues I had with thermald. Maybe some other software is setting a throttle?

Higher CPU temperature limits will not just allow you to run it faster but also to run it fast for longer periods.