erpalma/throttled #80: Provide ebuild for Gentoo

tbart, 2018-12-04:
From reading the reddit thread, it seems @erpalma is a Gentoo user. Why not provide an ebuild to install this tool?

I did it already, and it works for me (though I do not use systemd, that part would need checking).

If it works for you as well, probably add it to the readme.md
I can also submit it to bugs.gentoo.org.

Github won't allow me to upload ebuilds here, so here they are: https://gist.github.com/tbart/1c34657b3c19fb02f819ea020787af83
The init script obviously belongs to files/

bemeurer, 2018-12-12:
Gentoo user here, and ebuild would be a lifesaver!

henning-schild, 2018-12-17:
i also have an ebuild for that, might open a PR against the portage tree, or maybe we can use this repo as a staging area for it

erpalma, 2018-12-18:
Thanks a lot! :)

How about systemd? We might add a USE flag for that

erpalma, 2018-12-19:
Absolutely! Please do a PR, I'm testing your ebuild right now.

bemeurer, 2019-01-02:
Updates on this? An ebuild would be sweet!

henning-schild, 2019-01-17:
I do have an ebuild and could add it here. But in order to really use it people would have to copy it out or this repo would need to become a layman overlay as well. Maybe if @erpalma would start an empty overlay in his namespace i could add it there, because we would not want the actual code to be part of the overlay. Maybe an overlay can be a subdirectory of a git repo ... not sure.

erpalma, 2019-01-22:
Done!

closed 2019-01-22 by erpalma