erpalma/throttled #84: match supported machines before doing anything

henning-schild, 2018-12-17:
At the moment this script just seems to go ahead and mess around with machine specifics, that could go truly wrong if executed on a machine that is not supported. Those mmio regions might be backed by something we better not access.
I wrote an ebuild for gentoo (#80) but that can not go upstream with some DMI matching or alike.

erpalma, 2018-12-18:
I see what you means, but right now we do not have the full list of "supported" hardware. In the readme it is clear that if you try the tool on your machine, then you are also taking a little "risk". I might add some DMI matching, but I think that could give users a false sense of safety: I can match your T480s, but I don't know what exactly might go wrong if you have a newer revision or a discrete dGPU. We are already checking for the support of some features (e.g. temperature, cTDP), so I guess that the right solution to this problem is to perform more tests on the features advertised by the CPU and, w.r.t. the MCHBAR, I can check if the value is somewhat consistent to what I expect. I need to investigate further on this.

Tl;dr You are right, but I think that the solution should be a bit more general.

DEvil0000, 2018-12-18:
> In fact the whole thing should probably become a platform driver and be done inside the kernel.
>
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/erpalma/lenovo-throttling-fix/issues/84#issuecomment-448321921",
> "url": "https://github.com/erpalma/lenovo-throttling-fix/issues/84#issuecomment-448321921",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

True, true

erpalma, 2018-12-20:
I investigated a bit on the issue. Actually the CONFIG_TDP (offset 59A0) MCH register is available from gen 4 onward. Maybe I can just check the microarchitecture and add a `--force` flag to bypass the check for the expert user. Do you agree?

erpalma, 2018-12-27:
Please have a look at the testing branch!