erpalma/throttled #91: List of config files

otaj, 2019-01-13:
Hi,

first of all thanks for the amazing work, it makes the system usable. However, I believe most people like me are not very well versed in undervolting, MBR, etc. terminology. It would be nice to have either a wiki page or list of known config files that work well with the specific configuration of various laptops so that new users don't have to test it themselves.

For example, I have a similar configuration to the one listed in the readme (T480, MX150 dGPU, i7-8550U), however, because of the difference in having dGPU, I had to tweak a few values in my config file in order to make it usable.

c-edw, 2019-02-15:
I would be glad to share my undervolting config with you. I also have an i7-8550U so there's a good chance it will work for you too.

https://gist.github.com/c-edw/1dbb4a601c0cb7f628741ddf10dec74d

xmaysonnave, 2019-05-03:
I agree.  I'm fighting too with this throttle issue. 
I own a T470 with a i5-7300u.
I currently use the default configuration and I'm pretty scared to play with undervoltage.
This project has a wiki. It could be nice to publish suitable configurations experimented by users there.
Thanks for your great project.

erpalma, 2019-05-03:
Actually playing with undervoltage is quite safe, at least for the hardware. The only issue is that you might corrupt your filesystem due to crashes. Usually I do this from a live usb, but I guess you can do it from your distro and keep a `watch sync` in background. I use prime95 to test the stability of the system and gradually lower the voltage at 5 or 10 mV steps. When you reach the limit you set a conservative value by about 10/20 % just to play safe. During tests I also stop and re-launch prime95 because often crashes happens during transitions from low to high freq.

otaj, 2019-05-03:
I actually only recently found a script, which was testing safe PHC (earlier method of undervolting for older intel CPUs) undervolting by prime95 (https://searchcode.com/file/48898279/mprime-phc-setup) - I plan to port it to Python and undervolting by writing to MSR registers, I will share it here when it is done.

kakawait, 2019-08-29:
I'm highly interested for config running with `i5-i8350U` (t480s) if someone has. Otherwise I'll try myself I post here my result if succeed

mike391, 2019-12-20:
> I'm highly interested for config running with `i5-i8350U` (t480s) if someone has. Otherwise I'll try myself I post here my result if succeed

Any updates? Im also running an i5-8350U

sammyspark, 2020-01-01:
@kakawait Have you gone ahead and tried some voltages yet? I'm also running an i5-8350U t480s, and if you haven't tried yet I'd be glad to share my results.

kakawait, 2020-01-01:
@sammyspark sure I'm interested. I did some tweaks that seems to fix throttle and still stable but not sure if most optimal values. I will post my values as soon as I've my laptop on hands