erpalma/throttled #92: Single-core performance on T480s does not reach full potential

zimmski, 2019-01-13:
I ran Geekbench4 on a T480s with an i7-8550U (the same as in the README). Results are here https://browser.geekbench.com/v4/cpu/11625783 and I am surprised that I do not reach the same single-core performance as described in the README. 5391 vs 3757 is a **huge** difference. Hence, I am wondering what I am missing/how I can debug my setup to get to the same performance.

Setup:
- openSUSE Tumbleweed (latest revision)
- lenovo-throttling-fix (master which is f9c7f515493887cc4b4c122183524803fb942859) is enabled and status is all green
- HWP_Mode set to "True"
- Secure Boot is disabled in the BIOS
- Already rebooted

Looking at s-tui during the benchmarks tells me that the CPU is reaching 4.0GHz.

erpalma, 2019-01-18:
Just tested again: `https://browser.geekbench.com/v4/cpu/11692843`

Can you boot an ubuntu live and test again? Maybe this is something related to your distro.

zimmski, 2019-04-15:
Sorry for taking so long. I have a huge backlog on GitHub mails, I do not how to work through them at the moment.

I am reading the Tumbleweed mailing list and somebody wrote something related to this problem: most likely it is this https://lists.opensuse.org/opensuse-factory/2019-04/msg00126.html 

My score went up to 4488 with  "spectre_v2=retpoline,generic" as additional kernel parameters.