erpalma/throttled #93: Fix doesn't work anymore on t480s kernel 4.20.0

arudat, 2019-01-15:
Hi, I cannot start the script anymore because of that error message 
"lenovo_fix.py[13020]: [E] No valid Sysfs_Power_Path found!"

because of kernel update?

Thanks

erpalma, 2019-01-15:
Kernel 4.20 here, `/sys/class/power_supply/AC/online` present and no errors reported. You can try forcing `Sysfs_Power_Path: /sys/class/power_supply/AC/online` in the config file.

arudat, 2019-01-15:
Hey, I changed it in /etc/lenovo_fix.conf and hardcoded in the py script, no effect :/

erpalma, 2019-01-15:
conf parsing is done by `configparser` module. Can you please paste here the non-working `/etc/lenovo_fix.conf`?

DEvil0000, 2019-01-15:
does the user who executes the script have access to it? is the path correct?those sys pathes are kernel created so a kernel update might cause a change but its unlikely.

arudat, 2019-01-15:
@erpalma 
`[GENERAL]                                                                                                                                                                                                                                                  
# Enable or disable the script execution
Enabled: True                                                                                                                                                                                                                                              
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path = "/sys/class/power_supply/AC/online"

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 29
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 35
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 85
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 5
# Max package power for time window #1
PL1_Tdp_W: 44
# Time window #1 duration
PL1_Duration_s: 28
# Max package power for time window #2
PL2_Tdp_W: 44
# Time window #2 duration
PL2_Duration_S: 0.002
# Max allowed temperature before throttling
Trip_Temp_C: 95
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: False
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

[UNDERVOLT]
# CPU core voltage offset (mV)
CORE: 0
# Integrated GPU voltage offset (mV)
GPU: 0
# CPU cache voltage offset (mV)
CACHE: 0
# System Agent voltage offset (mV)
UNCORE: 0
# Analog I/O voltage offset (mV)
ANALOGIO: 0
`

@DEvil0000 root is the user and also has the right access, so anyway the script is recognized, otherwise   to disable the option in the config file would be ignored

erpalma, 2019-01-16:
`Sysfs_Power_Path = "/sys/class/power_supply/AC/online"`

should be

`Sysfs_Power_Path: "/sys/class/power_supply/AC/online"`

arudat, 2019-01-16:
@erpalma makes sense, but still get the same error

erpalma, 2019-03-11:
Now I see. If you get `[E] No valid Sysfs_Power_Path found!` then you are running an older version of the tool. In the latest you should see `[W] No valid Sysfs_Power_Path found! Trying upower method #1`.

closed 2019-03-14 by arudat