erpalma/throttled #94: Confirm working on Dell XPS15 9550 (i5-6300HQ)

adrfantini, 2019-01-20:
I can confirm this program is working on my Dell XPS15 9550 (i5-6300HQ) with the following configuration:
```
[GENERAL]
# Enable or disable the script execution
Enabled: True
# SYSFS path for checking if the system is running on AC power
Sysfs_Power_Path: /sys/class/power_supply/AC*/online

## Settings to apply while connected to Battery power
[BATTERY]
# Update the registers every this many seconds
Update_Rate_s: 30
# Max package power for time window #1
PL1_Tdp_W: 45
# Time window #1 duration
PL1_Duration_s: 30
# Max package power for time window #2
PL2_Tdp_W: 60
# Time window #2 duration
PL2_Duration_S: 0.005
# Max allowed temperature before throttling
Trip_Temp_C: 80
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

## Settings to apply while connected to AC power
[AC]
# Update the registers every this many seconds
Update_Rate_s: 10
# Max package power for time window #1
PL1_Tdp_W: 55
# Time window #1 duration
PL1_Duration_s: 45
# Max package power for time window #2
PL2_Tdp_W: 65
# Time window #2 duration
PL2_Duration_S: 0.007
# Max allowed temperature before throttling
Trip_Temp_C: 90
# Set HWP energy performance hints to 'performance' on high load (EXPERIMENTAL)
HWP_Mode: True
# Set cTDP to normal=0, down=1 or up=2 (EXPERIMENTAL)
cTDP: 0

[UNDERVOLT.BATTERY]
# CPU core voltage offset (mV)
CORE: -160
# Integrated GPU voltage offset (mV)
GPU: -100
# CPU cache voltage offset (mV)
CACHE: -160
# System Agent voltage offset (mV)
UNCORE: -50
# Analog I/O voltage offset (mV)
ANALOGIO: -50

[UNDERVOLT.AC]
# CPU core voltage offset (mV)
CORE: -120
# Integrated GPU voltage offset (mV)
GPU: -80
# CPU cache voltage offset (mV)
CACHE: -120
# System Agent voltage offset (mV)
UNCORE: 0
# Analog I/O voltage offset (mV)
ANALOGIO: 0
```

This device also works OK with the similar tool `intel-undervolt` and usually retains settings from Intel XTU on Windows even under Linux, in my experience.

With `throttled` enabled, I get `geekbench4` scores of 4073/11110 (single/multi). Disabled (using `balance_performance`), I get 3928/10393. I only tested over two runs so the variability in the performance could be large from run to run, I do not know.

adrfantini, 2019-01-21:
Actually, it appears that, at least on my Arch OS, the `/sys/class/power_supply/AC*/online` file is not created at boot IF booting from battery. Instead, `/sys/class/power_supply/BAT0` is created, and `throttled` fails because of this. `/sys/class/power_supply/AC*/online` is created as soon as AC is plugged in.

adrfantini, 2019-01-23:
Do you know what I need to change in the AUR pkgbuild to test the testing branch out?

erpalma, 2019-01-23:
Merged in master yesterday!

adrfantini, 2019-01-23:
Thanks. Seems to be working, I am getting `No valid Sysfs_Power_Path found! Trying upower method #1`. Maybe these warnings should be forwarded to `systemd`, so they are available in the logs.

erpalma, 2019-01-24:
Fixed in 56a91aa.

adrfantini, 2019-01-24:
Sweet, thanks! Closing?