erpalma/throttled #97: Setting IccMax?

arter97, 2019-02-08:
For most 8th Gen laptops I've seen, IccMax is still a big factor on why the processor is throttling. (And I think lifting up IccMax throttling should be the next milestone for this project as well.)

As an example, the LG Gram(8250U), suffers from current limit throttling if IccMax is left as default.
This prevents hitting 3.4 GHz on all cores, even after setting HWP bias to "performance".

On Windows, using ThrottleStop easily bypasses this issue.
On Linux, giving load to all cores throttles the frequency to 3.1 GHz immediately.

I've done a fair bit of research, but I could not find any solid information on setting IccMax via MSR register or MMIO access. The ThrottleStop author seems to have concluded the same, but recently updated the program with support for modifying IccMax.

I think contacting the ThrottleStop author is our best chance to do a similar thing on Linux.

I do not have write permissions to forum.notebookreview.com as of now, so if anyone could write on the forum and ask how he modified IccMax values, it'd be greatly appreciated.

Thanks!

DEvil0000, 2019-02-09:
you never now if a current limit is there for a reason (and it most likely is). changeing this will put the laptop in danger for hardware damage (short or longterm) or in the worst case catching fire.I can think of a ton of different reasons for setting a current limit.

arter97, 2019-02-09:
So is every other throttling, though.

DEvil0000, 2019-02-09:
thats not true. more current has way more danger.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/erpalma/throttled/issues/97#issuecomment-462021959",
"url": "https://github.com/erpalma/throttled/issues/97#issuecomment-462021959",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

arter97, 2019-02-09:
Well, that's what I thought https://github.com/erpalma/throttled/blob/master/README.md#disclaimer was for.

Nvm, got a reply from the ThrottleStop's author.

Thanks.

ejgallego, 2019-02-09:
> Nvm, got a reply from the ThrottleStop's author.

Could you share what the reply was? Thanks!

arter97, 2019-02-10:
Figured it out.

On Windows:
Wrmsr(0x150, dwIccMax, 0x80000017 | (dwAgent << 8));
dwAgent is 0 for CPU Core, 1 for iGPU and 2 for CPU Cache.
Command 17 is to write IccMax to the register and command 16 is how you read IccMax from the register.

0x3ff is the max on my 8250U(this would be likely for every other CPUs), so I'm using the following commands:
wrmsr -a 0x150 0x80000017000003ff
wrmsr -a 0x150 0x80000117000003ff
wrmsr -a 0x150 0x80000217000003ff

ejgallego, 2019-02-10:
Thanks @arter97 , is there some more information about the risks of current modification?

arter97, 2019-02-11:
There's a plenty of discussion of that on the ThrottleStop's thread.

You can use the search feature on that forum to find more information.

erpalma, 2019-02-11:
I can implement this in the tool and I might add a big warning when IccMax is overridden. 

Clearly the risk of raising this value is a physical damage both to the voltage regulators and the CPU either due to higher temperatures or currents above the absolute maximum ratings of the component.

erpalma, 2019-03-05:
I'm still missing the formula for calculate current given the value in the register (and vice versa)

erpalma, 2019-03-06:
Added in master!

closed 2019-03-06 by erpalma