erpalma/throttled #98: Not working on fedora 29

mdtita, 2019-02-10:
Installed all the dependencies, and that's the result of running install.sh

```
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
Building virtualenv...
Requirement already satisfied: configparser==3.5.0 in ./venv/lib/python3.7/site-packages (from -r requirements.txt (line 1)) (3.5.0)
Requirement already satisfied: dbus-python==1.2.8 in ./venv/lib/python3.7/site-packages (from -r requirements.txt (line 2)) (1.2.8)
Collecting PyGObject==3.30.4 (from -r requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/8c/1f/76533985b054473ef6ab1ba4d9c00d62da502f8b43d3171ae588ec81ae93/PyGObject-3.30.4.tar.gz
Collecting pycairo>=1.11.1 (from PyGObject==3.30.4->-r requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/a6/54/23d6cf3e8d8f1eb30e0e58f171b6f62b2ea75c024935492373639a1a08e4/pycairo-1.18.0.tar.gz
Installing collected packages: pycairo, PyGObject
  Running setup.py install for pycairo ... error
    Complete output from command /opt/lenovo_fix/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-l0i3cg8x/pycairo/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-5ulfblow/install-record.txt --single-version-externally-managed --compile --install-headers /opt/lenovo_fix/venv/include/site/python3.7/pycairo:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.7
    creating build/lib.linux-x86_64-3.7/cairo
    copying cairo/__init__.py -> build/lib.linux-x86_64-3.7/cairo
    copying cairo/__init__.pyi -> build/lib.linux-x86_64-3.7/cairo
    copying cairo/py.typed -> build/lib.linux-x86_64-3.7/cairo
    running build_ext
    Package x11 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `x11.pc'
    to the PKG_CONFIG_PATH environment variable
    Package 'x11', required by 'xrender', not found
    Package 'x11', required by 'cairo', not found
    Package 'x11', required by 'xext', not found
    Command '['pkg-config', '--cflags-only-I', 'cairo']' returned non-zero exit status 1.
    
    ----------------------------------------
Command "/opt/lenovo_fix/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-l0i3cg8x/pycairo/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-5ulfblow/install-record.txt --single-version-externally-managed --compile --install-headers /opt/lenovo_fix/venv/include/site/python3.7/pycairo" failed with error code 1 in /tmp/pip-install-l0i3cg8x/pycairo/

```

erpalma, 2019-02-11:
You do not have either pkg-config or cairo installed I guess.

mdtita, 2019-02-11:
```
 dnf install python3-cairo-devel cairo-gobject-devel gobject-introspection-devel dbus-glib-devel python3-devel make
Last metadata expiration check: 0:00:32 ago on Mon 11 Feb 2019 11:07:05 AM CET.
Package python3-cairo-devel-1.17.1-2.fc29.x86_64 is already installed.
Package cairo-gobject-devel-1.16.0-3.fc29.x86_64 is already installed.
Package gobject-introspection-devel-1.58.3-1.fc29.x86_64 is already installed.
Package dbus-glib-devel-0.110-3.fc29.x86_64 is already installed.
Package python3-devel-3.7.2-4.fc29.x86_64 is already installed.
Package make-1:4.2.1-10.fc29.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
⋊> ~ dnf install python3-cairo                                                                    11:08:33
Last metadata expiration check: 0:01:30 ago on Mon 11 Feb 2019 11:07:05 AM CET.
Package python3-cairo-1.17.1-2.fc29.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
⋊> ~ dnf install cairo-gobject                                                                    11:08:48
keybase                                                                    79 kB/s | 3.3 kB     00:00    
Package cairo-gobject-1.16.0-3.fc29.x86_64 is already installed.
Package cairo-gobject-1.16.0-3.fc29.i686 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
⋊> ~ dnf install gobject-introspection                                                            11:09:13
Last metadata expiration check: 0:00:22 ago on Mon 11 Feb 2019 11:08:53 AM CET.
Package gobject-introspection-1.58.3-1.fc29.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
⋊> ~ dnf install  pkg-config                                                                      11:09:17
Last metadata expiration check: 0:01:00 ago on Mon 11 Feb 2019 11:08:53 AM CET.
Package pkgconf-pkg-config-1.5.3-2.fc29.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!

```

mdtita, 2019-02-12:
actually fixed by installing `libX11-devel.x86_64`, should be added to Fedora requirements.

closed 2019-02-12 by mdtita