erpalma/throttled #100: Automatic undervolting configuration?

l29ah, 2019-02-13:
It would be really nice for this tool to be able to load the system and tune the voltages appropriately like https://bbs.archlinux.org/viewtopic.php?pid=1141702#p1141702 does for older Intel CPUs.

Meanwhile, thanks a lot for the tool and the ebuild. It works great on my 51nb X210!

c-edw, 2019-02-15:
That looks interesting. When you say 'older Intel CPUs', does that mean it doesn't work for modern ones? Do you happen to know how it tests stability without locking it up as it would if you undervolt too much?

erpalma, 2019-02-15:
This is a delicate topic. From my experience the system *will* freeze while identifying the lowest undervolt. Moreover while my system was rock solid under full load (prime95) for hours, when I stopped the stress test the system froze. Undervolt should be tested under full load with minimum and maximum frequencies, during transients and also while idle.

l29ah, 2019-02-15:
Yes.
The aforementioned script never locked up anything for me, as it brings up the voltage as soon as the test load fails to verify its results.

l29ah, 2019-02-15:
It tests the CPU correctness for every frequency step (and configures the voltage offset for every step).

nariox, 2019-02-16:
as erpalma mentioned, it is trickier to achieve this with modern CPUs. My guess is that we now control just the global voltage offset, not the voltages for each frequency.

A particular offset might be stable for 3.6GHz, 3.5GHz, but not for 3.3 GHz, etc. And the temperature also contributes to it. So an offset would have to be tested for all these frequencies (my processor report 33 possible frequencies with `cat /sys/devices/system/cpu/intel_pstate/num_pstates`) at high temperatures (e.g. if your processor goes from 3.6GHz to 400MHz while at 95C).

The link to the script is broken, but if I remember right, it added +5 at the end of the test to "ensure" stability, is that right?

l29ah, 2019-02-16:
On Sat, Feb 16, 2019 at 04:23:48AM -0800, nariox wrote:
> The link to the script is broken, but if I remember right, it added +5 at the end of the test to "ensure" stability, is that right? 

AFAIR it added 2, not sure what were the units though.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

otaj, 2019-05-10:
The code for the script can be found on the net, for example here. https://searchcode.com/file/48898279/mprime-phc-setup