erpalma/throttled #104: The service stoped on Fedora 29.

WongRuoran, 2019-02-27:
The install script can't run normally, so I installed it by running commands in the install.sh manually. But the service can't work.

`● lenovo_fix.service - Stop Intel throttling
   Loaded: loaded (/etc/systemd/system/lenovo_fix.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Wed 2019-02-27 15:51:35 CST; 1s ago
  Process: 7275 ExecStart=/opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py (code=exited, status=1/FAILURE)
 Main PID: 7275 (code=exited, status=1/FAILURE)

Feb 27 15:51:35 localhost.localdomain python3[7275]:     **kwargs)
Feb 27 15:51:35 localhost.localdomain python3[7275]:   File "/usr/lib64/python3.7/configparser.py", line 802, in _get
Feb 27 15:51:35 localhost.localdomain python3[7275]:     return conv(self.get(section, option, **kwargs))
Feb 27 15:51:35 localhost.localdomain python3[7275]:   File "/usr/lib64/python3.7/configparser.py", line 780, in get
Feb 27 15:51:35 localhost.localdomain python3[7275]:     d = self._unify_values(section, vars)
Feb 27 15:51:35 localhost.localdomain python3[7275]:   File "/usr/lib64/python3.7/configparser.py", line 1146, in _unify_values
Feb 27 15:51:35 localhost.localdomain python3[7275]:     raise NoSectionError(section) from None
Feb 27 15:51:35 localhost.localdomain python3[7275]: configparser.NoSectionError: No section: 'UNDERVOLT'
Feb 27 15:51:35 localhost.localdomain systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=1/FAILURE
Feb 27 15:51:35 localhost.localdomain systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.`

c-edw, 2019-03-01:
Does `/etc/lenovo_fix.conf` exist?

```
ls -l /etc/lenovo_fix.conf
```

If so, what does it contain?

I run Fedora 29 and the tool is running smoothly.

erpalma, 2019-03-02:
This is strange. I get `UNDERVOLT` section from config at line 283 only and a fallback is specified if the section or value is not found. Are you sure you are running the latest version?

WongRuoran, 2019-03-03:
> Does `/etc/lenovo_fix.conf` exist?
> 
> ```
> ls -l /etc/lenovo_fix.conf
> ```
> 
> If so, what does it contain?
> 
> I run Fedora 29 and the tool is running smoothly.

Yes it does. And [this is it's content.](https://paste.fedoraproject.org/paste/EvMTSj3KyYKc034l6o42GA)

WongRuoran, 2019-03-03:
> This is strange. I get `UNDERVOLT` section from config at line 283 only and a fallback is specified if the section or value is not found. Are you sure you are running the latest version?

Yes I am sure. I cloned it a few days ago.

WongRuoran, 2019-03-03:
> Does `/etc/lenovo_fix.conf` exist?
> 
> ```
> ls -l /etc/lenovo_fix.conf
> ```
> 
> If so, what does it contain?
> 
> I run Fedora 29 and the tool is running smoothly.

I tried to run install.sh but it read:

`./install.sh: line 10: dirname: command not found
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
./install.sh: line 20: cp: command not found
`
I copied everything in /usr/bin to /usr/sbin but this problem still exists.

c-edw, 2019-03-03:
What's the output of `echo $PATH`?

erpalma, 2019-03-03:
Can you please run the script in a shell and post the full traceback?

WongRuoran, 2019-03-04:
> What's the output of `echo $PATH`?

`
/home/ruoran/TeXlive/bin/x86_64-linux:/usr/share/Modules/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/home/ruoran/bin

`

WongRuoran, 2019-03-04:
> Can you please run the script in a shell and post the full traceback?

`./install.sh: line 10: dirname: command not found`
` Copying config file... Config file already exists, skipping.`
` Copying systemd service file...` 
`./install.sh: line 20: cp: command not found `

erpalma, 2019-03-04:
Sorry, my fault. I meant `lenovo_fix.py`.

WongRuoran, 2019-03-04:
> Sorry, my fault. I meant `lenovo_fix.py`.

```
  Traceback (most recent call last):
  File "./lenovo_fix.py", line 521, in <module>
    main()
  File "./lenovo_fix.py", line 465, in main
    config = load_config()
  File "./lenovo_fix.py", line 286, in load_config
    value = config.getfloat('UNDERVOLT', plane)
  File "/usr/lib64/python3.7/configparser.py", line 823, in getfloat
    fallback=fallback, **kwargs)
  File "/usr/lib64/python3.7/configparser.py", line 808, in _get_conv
    **kwargs)
  File "/usr/lib64/python3.7/configparser.py", line 802, in _get
    return conv(self.get(section, option, **kwargs))
  File "/usr/lib64/python3.7/configparser.py", line 780, in get
    d = self._unify_values(section, vars)
  File "/usr/lib64/python3.7/configparser.py", line 1146, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'UNDERVOLT'
```

erpalma, 2019-03-04:
Have a look at line 286 in master. Clearly you have an older version of the tool installed.

WongRuoran, 2019-03-04:
> Have a look at line 286 in master. Clearly you have an older version of the tool installed.

This is line 286:
`        value = config.getfloat('UNDERVOLT', plane)`
But I don't know what it means.
Pip in the "venv" directory is too old, but things didn't get better after I have updated it.

erpalma, 2019-03-04:
You are running an older version of the tool. I guess your `/bin/sh` is doing something weird with the path since you are experiencing `./install.sh: line 20: cp: command not found`. Try changing `/bin/sh` to `/bin/bash`.

WongRuoran, 2019-03-04:
> You are running an older version of the tool. I guess your `/bin/sh` is doing something weird with the path since you are experiencing `./install.sh: line 20: cp: command not found`. Try changing `/bin/sh` to `/bin/bash`.

I cloned the latest version and changed `/bin/sh` to `/bin/bash` but all of these don't work.The same problem occurred.

erpalma, 2019-03-04:
You can either manually copy/update the tool or prepend all the commands in the install script with their full paths or explicitly set PATH at the top of the script. Still, I think your distro has some issues since other people with F29 are running fine.

WongRuoran, 2019-03-04:
> You can either manually copy/update the tool or prepend all the commands in the install script with their full paths or explicitly set PATH at the top of the script. Still, I think your distro has some issues since other people with F29 are running fine.

I added `export PATH=/usr/bin:$PATH` to the top of "install.sh" and installed it without any problem. But the service still can't work.

```
● lenovo_fix.service - Stop Intel throttling
   Loaded: loaded (/etc/systemd/system/lenovo_fix.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Mon 2019-03-04 19:12:41 CST; 1s ago
  Process: 13728 ExecStart=/opt/lenovo_fix/venv/bin/python3 /opt/lenovo_fix/lenovo_fix.py (code=exited, status=1/FAILURE)
 Main PID: 13728 (code=exited, status=1/FAILURE)

Mar 04 19:12:41 localhost.localdomain systemd[1]: Started Stop Intel throttling.
Mar 04 19:12:41 localhost.localdomain python3[13728]: Traceback (most recent call last):
Mar 04 19:12:41 localhost.localdomain python3[13728]:   File "/opt/lenovo_fix/lenovo_fix.py", line 6, in <module>
Mar 04 19:12:41 localhost.localdomain python3[13728]:     import dbus
Mar 04 19:12:41 localhost.localdomain python3[13728]: ModuleNotFoundError: No module named 'dbus'
Mar 04 19:12:41 localhost.localdomain systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=1/FAILURE
Mar 04 19:12:41 localhost.localdomain systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.
```

WongRuoran, 2019-03-04:
> You can either manually copy/update the tool or prepend all the commands in the install script with their full paths or explicitly set PATH at the top of the script. Still, I think your distro has some issues since other people with F29 are running fine.

It reads `[I] Detected CPU architecture: Intel Kaby Lake (R)` when I tried to run `sudo python3 lenovo_fix.py`

erpalma, 2019-03-04:
Can you do a `pip freeze` from within the virtualenv?

WongRuoran, 2019-03-04:
> Can you do a `pip freeze` from within the virtualenv?

It runs without output.
I found that I can't `import dbus` using `/opt/lenovo_fix/venv/bin/python3` but I can do it with `/usr/bin/python3`. 
The following is output of `ls -l /opt/lenovo_fix/venv/bin/python3`:
`lrwxrwxrwx. 1 root root 16 Mar  4 19:06 python3 -> /usr/bin/python3`

erpalma, 2019-03-04:
Your virtualenv is empty, that's why you can't find the dbus module. Just do a `pip install -r requirements.txt`

WongRuoran, 2019-03-04:
> Your virtualenv is empty, that's why you can't find the dbus module. Just do a `pip install -r requirements.txt`

I successfully installed them but it can't import dbus still.
```
>>> import dbus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'dbus'
```

WongRuoran, 2019-03-13:
The problem can be solved in this way.
https://github.com/erpalma/throttled/issues/106#issue-419590290

closed 2019-03-13 by WongRuoran