erpalma/throttled #111: Undervolting does not work on t480s.

youbuntu, 2019-04-16:
My laptop specifications
OS: Pop!_OS 18.10 x86_64
Shell: bash 4.4.19 
CPU: Intel i5-8250U (8) @ 3.400GHz
GPU: Intel UHD Graphics 620 
Memory: 2730MiB / 15798MiB

I modified the undervolting value in lenovo_fix.conf and saved it, but it does not apply. I opened the lenovo_fix.conf confirmed that the modification has been completed successfully. However, when I checked with ./lenovo_fix.py --monitor, the undervolting value was not changed. I have tried sudo ./install.sh and sudo systemctl restart lenovo_fix.service but it still does not apply. 

Please help me how to solve it.

erpalma, 2019-04-16:
Can you check with `--debug`?

youbuntu, 2019-04-16:
My configuration file is as below.
lenovo_fix.conf : https://pastebin.com/BUk8nEV6
lenovo_fix --monitor : https://pastebin.com/AdhQd0QR
lenovo_fix --debug : https://pastebin.com/bA78pcev

erpalma, 2019-04-16:
Are you sure that you are editing `/etc/lenovo_fix.conf`?

youbuntu, 2019-04-16:
Oh my God ... I was editing the file in /home/lenovo-throttling-fix/etc/lenovo_fix.conf. As you said, I modified the file in /etc/lenovo_fix.conf so undervolving was applied normally. I hope you understand my mistake as a beginner to Linux. Thank you very much for your help.

closed 2019-04-16 by erpalma