erpalma/throttled #112: HP spectre x360 not working, help!

caoHenriques, 2019-04-18:
Hi,

congratulations for this amazing tool :)
I have an hp spectre x360 with i7 8550u. it is not possible to increase TDP in XTU or throuttlestop in Windows for long loads because this setting is locked, and there is no advanced options in BIOS related with TDP. 

Could this tool work in my case? If I could get it working in linux I will port it to Windows, I'm sure a lot of people would benefit. Please help.

For instance, I configurated:
PL1_Tdp_W: 16       PL1_Duration_s: 900 
PL2_Tdp_W: 16       PL2_Duration_S: 0.002 

but after 70 sec, the power drops to 15W
as soon as I start the stress test, I get "Power: LIM" in --monitor
when the power drops down to 15W there is no difference in --monitor outputs, it seems that is not a thermal problem
if I configure 25W instead of 16 W, this throttling happens in just 30 sec.

Any ideas?
thank you very much

erpalma, 2019-04-18:
Peak temperatures?

caoHenriques, 2019-04-18:
thank you very much for your help. 
with 16w temperatures are 85ºC maximum
using --monitor, it only says Power: LIM, never thermal limited.
I can provide more information of the --monitor and --debug outputs, what do you need?

erpalma, 2019-04-18:
When you set 25W, can you tell me the peak temperature?

caoHenriques, 2019-04-19:
Hi, sorry for my late reply

with 25 W in PL1 and PL2:
after starting the stress test my temps rise up to 97ºC and then immediately power decreases down to 15W, but the Fan do not even start, the fan only start spinning after a while running the stress test when temps are already down (because it is at 15W)

so I run another test after the first one (since now fan was spinning) 
now power oscillated between 18W and 25W, temps were allways lower than 88ºC but after some seconds power was caped to 15W

I don't need 25W, probably my ultrabook cannot keep low temperatures during long runs at 25W, but I believe it can at 18 W or even 20W, because at 15W fans are not even at maximum speed

thanks for your help

erpalma, 2019-04-20:
The fan issue is related to the thermal compound of the heatsink, I guess you should consider replacing it.

caoHenriques, 2019-04-20:
really? I always thought that in these ultrabooks the fans start spinning only after a while of continuous heavy load.  
But using the 16 W or 17 W the same power throttling happens although temperatures are always low.
And "Power: LIM" appears in --monitor as soon I start the stress test even when temperatures are still around 50ºC.

So do you thing I should send it to warranty (I also have an insurance that should cover any damage or robbery)?

thanks

erpalma, 2019-04-22:
If your paste has low thermal conductivity, then your CPU die temperature will increase much faster than the temperature of the heatsink. I use liquid metal and the temperature rise is much better, but also with a good paste you should be fine.

My guess is that if your CPU hits some EC/UEFI defined temp threshold, then PL1/2 are forced to a low value.

toumorokoshi, 2023-03-05:
Just trying this out for myself, and I had the same experience.

> My guess is that if your CPU hits some EC/UEFI defined temp threshold, then PL1/2 are forced to a low value.

Wouldn't this still be overridden by the tool? IIUC when these limits are hit the EC forces the values to something lower, then `throttled` overrides them again.

I don't know if this is any help, but I found another experience even for windows here: https://www.reddit.com/r/overclocking/comments/t14b5a/is_it_possible_that_hp_disabled_power_limit/.

toumorokoshi, 2023-03-05:
Also an HP support post effectively refusing to allow this to be configurable in the BIOS: https://h30434.www3.hp.com/t5/Notebook-Software-and-How-To-Questions/TDP-locked-by-BIOS-at-15watt/td-p/7575889.

So it seems the solution here is to not buy HP next time and go with a manufacturer willing to give a little more freedom.