erpalma/throttled #113: Support for Solus distro (dependencies names)

shbach, 2019-04-23:
As described in [this reddit thread](https://www.reddit.com/r/SolusProject/comments/bfy4uk/help_with_intel_cpu_throttling_fix_on_solus/eljajwr/), the packages names in Solus are slightly different so the script may be failing to run due to this. 

configparser = python-configparser
dbus-python = python-dbus
PyGObject = python-gobject
setuptools = python-setuptools

Any help in getting this to work would be most appreciated!

criscodex, 2019-05-07:
I'm experiencing the same issue.

erpalma, 2019-05-08:
I have no experience with Solus, can you post the traceback or the exact error?

shbach, 2019-05-08:
Based on the instructions for the other distros I tried to find equivalent packages and installed these:
`gcc python3-devel dbus-glib-devel python3-cairo-devel libcairo-devel python3-gobject-devel`

When I run `sudo ./lenovo-throttling-fix/install.sh` this is the output:

```
Copying config file...
Config file already exists, skipping.
Copying systemd service file...
Building virtualenv...
Requirement already satisfied: configparser==3.5.0 in ./venv/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (3.5.0)
Collecting dbus-python==1.2.8 (from -r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/3f/e7/4edb582d1ffd5ac3c84188deea32e960b5c8c0fe1da56ce70224f85ce542/dbus-python-1.2.8.tar.gz
Collecting PyGObject==3.30.4 (from -r requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/8c/1f/76533985b054473ef6ab1ba4d9c00d62da502f8b43d3171ae588ec81ae93/PyGObject-3.30.4.tar.gz
Collecting pycairo>=1.11.1 (from PyGObject==3.30.4->-r requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/48/20/5e83af98eb897935bf7dc39455e892ba866feebb9b7c3b392982866f9958/pycairo-1.18.1.tar.gz
Installing collected packages: dbus-python, pycairo, PyGObject
  Running setup.py install for dbus-python ... error
    Complete output from command /opt/lenovo_fix/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-4tgp4777/dbus-python/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-n49k1oej/install-record.txt --single-version-externally-managed --compile --install-headers /opt/lenovo_fix/venv/include/site/python3.6/dbus-python:
    running install
    running build
    creating /tmp/pip-install-4tgp4777/dbus-python/build
    creating /tmp/pip-install-4tgp4777/dbus-python/build/temp.linux-x86_64-3.6
    checking whether make supports nested variables... no
    checking whether to enable maintainer-specific portions of Makefiles... no
    checking for gcc... gcc
    checking whether the C compiler works... no
    configure: error: in `/tmp/pip-install-4tgp4777/dbus-python/build/temp.linux-x86_64-3.6':
    configure: error: C compiler cannot create executables
    See `config.log' for more details
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-4tgp4777/dbus-python/setup.py", line 109, in <module>
        tests_require=['tap.py'],
      File "/opt/lenovo_fix/venv/lib/python3.6/site-packages/setuptools/__init__.py", line 143, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/opt/lenovo_fix/venv/lib/python3.6/site-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.6/distutils/command/install.py", line 545, in run
        self.run_command('build')
      File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-4tgp4777/dbus-python/setup.py", line 62, in run
        cwd=builddir)
      File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/tmp/pip-install-4tgp4777/dbus-python/configure', '--disable-maintainer-mode', 'PYTHON=/opt/lenovo_fix/venv/bin/python3', '--prefix=/tmp/pip-install-4tgp4777/dbus-python/build/temp.linux-x86_64-3.6/prefix']' returned non-zero exit status 77.
    
    ----------------------------------------
Command "/opt/lenovo_fix/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-4tgp4777/dbus-python/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-n49k1oej/install-record.txt --single-version-externally-managed --compile --install-headers /opt/lenovo_fix/venv/include/site/python3.6/dbus-python" failed with error code 1 in /tmp/pip-install-4tgp4777/dbus-python/
```

Based on the error messages I also installed:
`pip python-setuptools`
But still same error.
 
When I asked for help in the Solus subreddit this is what I was told:
"The issue is with the dependencies names, requirements.txt looks for

configparser==3.5.0

dbus-python==1.2.8

PyGObject==3.30.4

But we have python-configparser instead of configparser, python-dbus instead of dbus-python, python-gobject instead of PyGObject and python-setuptools instead of setuptools, to technically make it run you ll need to change the code itself replacing the package names :/"

If there is any other information I can provide please let me know.

erpalma, 2019-05-09:
The issue is clearly `checking whether the C compiler works... no`. Can you please include `config.log`?

shbach, 2019-05-09:
Sorry, I'm still very much a newbie. I've been trying but I still have no idea where config.log is located. Do you know where I might find that file?

criscodex, 2019-05-09:
It looks like the `/tmp` install dir is blown away when the install script exits, and the log file is deleted along with it.

shbach, 2019-05-09:
Oh well that explains why I couldn't find it in `/tmp`. I don't know how to stop it from deleting the file  though.

shbach, 2019-05-18:
I figured it out!

Basically needed to install the Solus equivalent of `build-essentials` by running:
`sudo eopkg install -c system.devel` 
See [here](https://getsol.us/articles/package-management/basics/en/#base-development-tools) for more info.

So for future reference anyone on Solus and trying to install the script run these commands:
```
sudo eopkg it -c system.devel
sudo eopkg it git python3-devel dbus-glib-devel python3-cairo-devel libcairo-devel python3-gobject-devel
git clone https://github.com/erpalma/lenovo-throttling-fix.git
sudo ./lenovo-throttling-fix/install.sh
```

@criscodex Let me know if this works for you.

criscodex, 2019-05-19:
@shbach thanks a bunch - that fixed it :)

DanMossa, 2020-04-15:
Would a PR be accepted for adding this to the README.md?

shbach, 2020-04-15:
I created a PR

PR #189 (2020-04-16): Update README.md
Added installation instructions for Solus. Closes #113
closed 2020-04-16 by erpalma