erpalma/throttled #116: Error in writemsr

dgllghr, 2019-05-13:
Dell XPS 9575
Intel i7-8705G
Fedora 30 (installed through https://copr.fedorainfracloud.org/coprs/abn/throttled/ so apologies if this is an issue with the package)

I am hitting the following error when running throttled as a systemd service. 

```
May 13 13:30:20 federation systemd[1]: Started Stop Intel throttling.
May 13 13:30:20 federation throttled[22393]: [I] Detected CPU architecture: Intel Kaby Lake (R)
May 13 13:30:20 federation throttled[22393]: [I] Loading config file.
May 13 13:30:20 federation throttled[22393]: [I] Starting main loop.
May 13 13:30:20 federation throttled[22393]: Exception in thread Thread-1:
May 13 13:30:20 federation throttled[22393]: Traceback (most recent call last):
May 13 13:30:20 federation throttled[22393]:   File "/usr/lib64/python3.7/threading.py", line 917, in _bootstrap_inner
May 13 13:30:20 federation throttled[22393]:     self.run()
May 13 13:30:20 federation throttled[22393]:   File "/usr/lib64/python3.7/threading.py", line 865, in run
May 13 13:30:20 federation throttled[22393]:     self._target(*self._args, **self._kwargs)
May 13 13:30:20 federation throttled[22393]:   File "/usr/bin/throttled", line 554, in power_thread
May 13 13:30:20 federation throttled[22393]:     writemsr(0x64B, write_value)
May 13 13:30:20 federation throttled[22393]:   File "/usr/bin/throttled", line 121, in writemsr
May 13 13:30:20 federation throttled[22393]:     raise e
May 13 13:30:20 federation throttled[22393]:   File "/usr/bin/throttled", line 112, in writemsr
May 13 13:30:20 federation throttled[22393]:     os.write(f, struct.pack('Q', val))
May 13 13:30:20 federation throttled[22393]: OSError: [Errno 5] Input/output error
May 13 13:30:30 federation systemd[1]: Stopping Stop Intel throttling...
May 13 13:30:30 federation systemd[1]: throttled.service: Main process exited, code=killed, status=15/TERM
May 13 13:30:30 federation systemd[1]: throttled.service: Succeeded.
```

Let me know if I can provide any other information that would be useful for tracking this down

erpalma, 2019-05-14:
Do you have secure boot enabled?

tobimat80, 2019-05-14:
I have the same issue with my DELL Precision 5530 (i9-8950HK). Secure boot is disabled. Running Ubuntu 18.04.2 LTS

erpalma, 2019-05-14:
Hmm strange. Does `/dev/cpu/0/msr` exists?

tobimat80, 2019-05-14:
Yes:

`crw------- 1 root root 202, 0 Mai 14 10:32 /dev/cpu/0/msr`

dgllghr, 2019-05-14:
I can confirm that my secure boot is also disabled and `/dev/cpu/0/msr` exists for me too

LoipesMas, 2019-08-28:
DELL Latitude e6540, same issue. Secure boot disabled, `/dev/cpu/0/msr` exists, Debian 10

erpalma, 2019-08-28:
Does `wrmsr` works?

LoipesMas, 2019-08-31:
`wrmsr 0x64b 0x80000000` results in `wrmsr: CPU 0 cannot set MSR 0x0000064b to 0x0000000080000000`, even after `modprobe msr`.

Firenox89, 2019-10-04:
I also own a Dell XPS 9575 running Arch on it.
The thing is not that I can't write to msr at all, but just not to `0x64b`.
Writing to `0x150` and `0x1a2` works fine.

Firenox89, 2019-10-04:
```
➜  ~ sudo rdmsr 0 0x64b                                                                           
[sudo] password for root: 
0x0000000080000000
```
If I read the Intel docu correctly, this value means that the TDP config is locked...
[image]