erpalma/throttled #118: Support for Lenovo Thinkpad T490s

efficiosoft, 2019-05-30:
Hi,

Thanks for this very useful fix! Too sad Lenovo doesn't get this implemented in an OS-independent way.

I tried running it on a T490s today, with no noticeable success. Power limit stays at 25W for a few seconds and then drops to 15W, where it stays with a CPU temp of 70°C.

On Windows, the TDP limit is 51W first and then stays at 25W.

Here's the debug output:
```
[I] Detected CPU architecture: Intel Kaby Lake (R)
[I] Loading config file.
[D] cpu platform info: maximum non turbo ratio = 20
[D] cpu platform info: maximum efficiency ratio = 4
[D] cpu platform info: minimum operating ratio = 4
[D] cpu platform info: feature ppin cap = 0
[D] cpu platform info: feature programmable turbo ratio = 1
[D] cpu platform info: feature programmable tdp limit = 1
[D] cpu platform info: number of additional tdp profiles = 2
[D] cpu platform info: feature programmable temperature target = 1
[D] cpu platform info: feature low power mode = 1
[D] core 0 thermal status: thermal limit status = 0
[D] core 0 thermal status: thermal limit log = 0
[D] core 0 thermal status: prochot or forcepr status = 0
[D] core 0 thermal status: prochot or forcepr log = 0
[D] core 0 thermal status: crit temp status = 0
[D] core 0 thermal status: crit temp log = 0
[D] core 0 thermal status: thermal threshold1 status = 0
[D] core 0 thermal status: thermal threshold1 log = 0
[D] core 0 thermal status: thermal threshold2 status = 0
[D] core 0 thermal status: thermal threshold2 log = 0
[D] core 0 thermal status: power limit status = 0
[D] core 0 thermal status: power limit log = 1
[D] core 0 thermal status: current limit status = 0
[D] core 0 thermal status: current limit log = 0
[D] core 0 thermal status: cross domain limit status = 0
[D] core 0 thermal status: cross domain limit log = 0
[D] core 0 thermal status: cpu temp = 40
[D] core 0 thermal status: temp resolution = 1
[D] core 0 thermal status: reading valid = 1
[D] core 1 thermal status: thermal limit status = 0
[D] core 1 thermal status: thermal limit log = 0
[D] core 1 thermal status: prochot or forcepr status = 0
[D] core 1 thermal status: prochot or forcepr log = 0
[D] core 1 thermal status: crit temp status = 0
[D] core 1 thermal status: crit temp log = 0
[D] core 1 thermal status: thermal threshold1 status = 0
[D] core 1 thermal status: thermal threshold1 log = 0
[D] core 1 thermal status: thermal threshold2 status = 0
[D] core 1 thermal status: thermal threshold2 log = 0
[D] core 1 thermal status: power limit status = 0
[D] core 1 thermal status: power limit log = 1
[D] core 1 thermal status: current limit status = 0
[D] core 1 thermal status: current limit log = 0
[D] core 1 thermal status: cross domain limit status = 0
[D] Undervolt plane CORE - write -130 mV (0xef600000) - read -130 mV (0xef600000) - match OK
[D] core 1 thermal status: cross domain limit log = 0
[D] core 1 thermal status: cpu temp = 37
[D] core 1 thermal status: temp resolution = 1
[D] core 1 thermal status: reading valid = 1
[D] core 2 thermal status: thermal limit status = 0
[D] core 2 thermal status: thermal limit log = 0
[D] core 2 thermal status: prochot or forcepr status = 0
[D] core 2 thermal status: prochot or forcepr log = 0
[D] core 2 thermal status: crit temp status = 0
[D] core 2 thermal status: crit temp log = 0
[D] core 2 thermal status: thermal threshold1 status = 0
[D] core 2 thermal status: thermal threshold1 log = 0
[D] core 2 thermal status: thermal threshold2 status = 0
[D] core 2 thermal status: thermal threshold2 log = 0
[... 94 lines cut ...]
[D] core 7 thermal status: thermal threshold2 log = 0
[D] core 7 thermal status: power limit status = 0
[D] core 7 thermal status: power limit log = 1
[D] Undervolt plane GPU - write -80 mV (0xf5c00000) - read -80 mV (0xf5c00000) - match OK
[D] core 7 thermal status: current limit status = 0
[D] core 7 thermal status: current limit log = 0
[D] core 7 thermal status: cross domain limit status = 0
[D] core 7 thermal status: cross domain limit log = 0
[D] core 7 thermal status: cpu temp = 41
[D] core 7 thermal status: temp resolution = 1
[D] core 7 thermal status: reading valid = 1
[D] Undervolt plane CACHE - write -130 mV (0xef600000) - read -130 mV (0xef600000) - match OK
[D] Undervolt plane UNCORE - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[D] TEMPERATURE_TARGET - write 0x5 - read 0x5 - match OK
[D] Undervolt plane ANALOGIO - write 0 mV (0x0) - read 0 mV (0x0) - match OK
[D] CONFIG_TDP_CONTROL - write 0x0 - read 0x0 - match OK
[D] MSR PACKAGE_POWER_LIMIT - write 0x9e80e000dc8198 - read 0x9e80e000dc8198 - match OK
[D] MCHBAR PACKAGE_POWER_LIMIT - write 0x9e80e000dc8198 - read 0x9e80e000dc8198 - match OK
[I] Starting main loop.
```

Do you think the T490s can be supported?

Best regards
Robert

erpalma, 2019-05-31:
I guess they are limiting the TDP to 15W in a new way on the T490 series. Maybe setting the correct UUID (#70) could help

efficiosoft, 2019-06-01:
Hi,

Setting the profile seems to help, but even when it's done directly at boot, before any CPU stressing has been performed. I put it into `/etc/rc.local` for that reason and now TDP always is 28W, no short term limit seems to be applied, even though I configured 51W for 29s via throttled.

I'm fine with the behaviour for now, 28W permanent is enough for thermal throttling to kick in after a few seconds, but it would be nice to have the 51W short term limit as on Windows.

efficiosoft, 2019-06-01:
BTW, they really did a bad job with the terrible cooling on T490s, the i7-7565U undervolted at -110mV stabilizes at only 4 x 2.6 GHz at 95°C after ten minutes.

erpalma, 2019-06-01:
Yes I know, you have a single heat pipe :/

HarryWindsor, 2019-09-15:
> Setting the profile seems to help, but only when it's done directly at boot, before any CPU stressing has been performed. I put it into `/etc/rc.local` for that reason and now TDP always is 28W, no short term limit seems to be applied, even though I configured 51W for 29s via throttled.

I have the same TDP issues with my T490. What do you mean by "Setting the profile" in `/etc/rc.local`, could you please explain this to me?

Danke

PHillner, 2019-09-16:
I also had throttling issues on my T490 (should be similar enough processor to the T490s). In addition to using the throttled daemon, I had to disable secure boot through BIOS for the throttling issue to go away.
Thought this may be a point some might find useful.

HarryWindsor, 2019-09-16:
@PHillner What BIOS version do you have installed? I am running N2IET74W (1.52) with "Secure Boot"="Disabled", "Platform Mode"="User Mode" and "Secure Boot Mode"="Standard Mode".

I get 100% CPU usage, but only ~11W Power usage at below 60°C.

PHillner, 2019-10-24:
@HarryWindsor Sorry for the late reply. I have N2IET76W (1.54) at this moment, which seems to be a different version than I had when I previously commented, as v1.54 was released 30 September 2019. This makes me think I had v1.52 back then.
At this moment, I am getting 3.5-4 W idle power usage with ~50 C temperature.