erpalma/throttled #120: Confirmed working on ThinkPad E590, this solves a major issue with RX 550X GPU

mp035, 2019-06-05:
I just want to confirm that this worked on my ThinkPad E590 with i7-8565U.  Prior to installing this fix, the CPU would be throttled to 200Mhz whenever an application was started that uses the AMD RX550X GPU.  Now it runs at a reasonable frequency.   I lowered the maximum AC powered temperature to 85 deg C as I thought 95 was a bit close to the limit.  Apart from that I did not make any config changes.  Thank you so much for your work on this script.  I will take some time to try to work out exactly why the GPU causes throttling and report back.

mp035, 2019-06-05:
It seems as though the e590 has MSR 0x1A2 set to 23 (100-23 = 77 deg C).  I think that starting the GPU forces the CPU package temperature above that threshold, causing the CPU to continuously throttle.  When I ran:
`sudo wrmsr -a 0x1a2 0xF000000`
In an attempt to raise the thermal limit to 85 deg C whilst the GPU was running,  the issue went away, and shortly returned.  Interestingly reading the same MSR after the controller had restored the values yields a value of 32 (100-32 = 68 deg C!) no wonder it throttles.  I am not sure why the value changes.

Either way this script definitely solves the problem.  Thanks again!

leonardblaschek, 2019-07-30:
I'll just butt in here to confirm that your fix works beautifully in my thinkpad E470 (i7-7500U/GeForce 940MX). Went from ~2ghz under load to ~3.2. Albeit slightly warmer ;)

Thanks for your work and putting it out here!

erpalma, 2019-07-30:
Thanks for the feedback :)

Dahdoul, 2019-08-13:
+1 Confirmed working on ThinkPad E590, i7 with RX 550X GPU

XeonBloomfield, 2019-09-01:
First of all I can thank you @erpalma for that project. I just bought ThinkPad E590 with i7 and RX 550X and I must confirm that it was completely unusable together. Would you consider adding it to officially supported platforms, because it works out of box with your project using current instructions even without modifying default configuration?

I have also a question for @mp035 and @Dahdoul about their final configuration. How did you configure P1, P2 and undervolting?

mp035, 2019-09-02:
Hi @XeonBloomfield , I would like to echo the thanks to @erpalma.  This laptop is wonderful, but would be **USELESS** without this script.  I left P1, P2, and undervolting at the project defaults.  The only modification I made to the config was to lower the AC powered temperature threshold to 85 deg C, but that is just me being overly cautious.

closed 2019-09-02 by mp035

erpalma, 2019-09-03:
Thank you for the feedback :)

theartful, 2019-10-27:
This works great on E490, with great performance boost compared to before. Thanks a lot for developing this!

kafran, 2019-11-19:
@theartful @mp035 I'm thinking about to buy the E490 with i7 and GPU but I'm afraid of these heating problem. What do you guys recommend? It seems Lenovo has release a BIOS update (1.14) for the previous version E480 which massively reduced the GPU performance to solve this heating problem. My fear is that this could happen to E490 too. Actually, the E490's 1.20 BIOS Update brings this: - (Important) Follow Thermal modify PL1/PL2 to 25W/29W. But I don't know what this mean. Thanks for your attention, I’m looking forward to your reply.

mp035, 2019-11-19:
Hi Kafran, I have had the E590 i7 with rx550x GPU for nearly 6 months now, and I have to say, as a Linux machine, I love it.  Please be aware though, this is not a gaming machine.  I use it for software development, and light gaming on Steam.  The GPU seems to handle all of this well but I am unable to get phoronix-test-suite to recognize the AMD GPU (even though glxinfo recognizes it) so I can't even give you a graphics benchmark.

kafran, 2019-11-20:
> Hi Kafran, I have had the E590 i7 with rx550x GPU for nearly 6 months now, and I have to say, as a Linux machine, I love it. Please be aware though, this is not a gaming machine. I use it for software development, and light gaming on Steam. The GPU seems to handle all of this well but I am unable to get phoronix-test-suite to recognize the AMD GPU (even though glxinfo recognizes it) so I can't even give you a graphics benchmark.
> 
> EDIT: Even though phoronix test-suite doesn't show the correct graphics card, radeontop inidicates that the discrete card is used, so for comparison, my very old XPS15 9530 (GeForce GT750M) got 31 fps on unigine-heaven@1024x768, but the thinkpad only averaged 28.75 fps. This however should improve with newer kernels because the thinkpad uses muxless gpu offloading, whereas the XPS uses switchable graphics.

Hi @mp035. Thank you for your reply. I have no experience with a GPU as I never had one. I use a Dell Inspiron with i5 4th gen. I'm thinking about to upgrade as I'm starting to miss modern ports (USB-C, GiB Ethernet, M.2 SSD, etc.). I'm a Linux user for the past 10+ years. I use the computer mainly for Python development and Data Science, I will be using the GPU mostly for small parallel processing. @mp035 have you updated the BIOS to the new 1.20? If so, how this impact the Thermal problem? Also, how this hibrid card thing works on Linux? I have no idea. For the past year I have been using the POP OS Linux Distro. I was even thinking about to buy a System76 laptop but it is too expensive for my budget. I know POP has custom features to deal with nVidia cards. I don't know what challenges may arise with AMD on Linux.

mp035, 2019-11-20:
@kafran, AMD GPU drivers on Linux are excellent.  Although performance is not as good as Nvidia right now, there is no need to restart the laptop to enable the discrete graphics (enabling GPU on Nvidia requires a restart).  Amdgpu drivers are in the Kernel, and if you want to use the GPU, you start your application with "DRI_PRIME=1 application".  That's it, everything just works.  If you are using this laptop for python development on Linux, I think you will be very happy.  My BIOS Version is R0YET31W (1.14).  I'm running Manjaro Linux and all hardware was detected correctly out of the box.

Also note that the latest Linux kernel has some major improvements in amdgpu drivers, so I expect to see performance improvements over time.

kafran, 2019-11-20:
I'm just waiting for the BlackFriday to see if prices gonna drop here in my country. @mp035, the 1.20 BIOS update brings the following information: "- (Important) Follow Thermal modify PL1/PL2 to 25W/29W". Do you know what this mean? I'm afraid BIOS update could cut GPU performance like what happened with E480.

theartful, 2019-11-20:
@kafran This is clearly not a gaming laptop, but gaming performance (on wine) is not good compared to my older laptop which had a weaker graphics card. The problem is that both CPU and dGPU overheat very quickly resulting in throttling. I end up restricting CPU clock speed to 1.8G (disabling turboboost), and also restricting dGPU clock speed using the amdgpu driver to have consistent gaming experience.

kafran, 2019-11-20:
> @kafran This is clearly not a gaming laptop, but gaming performance (on wine) is not good compared to my older laptop which had a weaker graphics card. The problem is that both CPU and dGPU overheat [...]

What do you think? Would you buy it with i7 + GPU? Does it worth it? Or should I get the i5 without GPU?

mp035, 2019-11-20:
@kafran, from what I can find on the net, the 1.20 bios update improves the performance by **raising** CPU TDP to 25/29W:
[https://forums.lenovo.com/t5/ThinkPad-11e-Windows-13-E-and/E590-with-Radeon-RX550X-CPU-power-inconsistent-after-reboot-Bug/td-p/4571503](https://forums.lenovo.com/t5/ThinkPad-11e-Windows-13-E-and/E590-with-Radeon-RX550X-CPU-power-inconsistent-after-reboot-Bug/td-p/4571503)
> First, I noticed that there is a BIOS update (Version 1.20) that should improve the performance of the CPU - I installed that, and indeed the expanded power limits improve the performance significantly (by about 30% on this machine). I also installed all the latest drivers for the machine, downloaded from the official page for this model on Lenovo site. That includes everything related to power management.

Please also note that the thermal bug that is being mentioned in the linked post is not present on linux thanks to erpalma's FANTASTIC work here.  It appears to be a bug in the windows equivalent of erpalma/throttled.

mp035, 2019-11-20:
@kafran, Update: After reading about the 1.20 bios update, I went ahead and installed it and re-ran unigine-heaven.  The test score improved from 28.75fps on 1.14 BIOS to 34.81fps on 1.20 BIOS.  I can conclusively say that 1.20 bios update does not hinder performance.

kafran, 2019-11-21:
> Please also note that the thermal bug that is being mentioned in the linked post is not present on linux thanks to erpalma's FANTASTIC work here. It appears to be a bug in the windows equivalent of erpalma/throttled.

Isn't _throttled_ already a "bug" correction?
Do you think a heatsink mod could solve this?

kafran, 2019-11-21:
> @kafran, Update: After reading about the 1.20 bios update, I went ahead and installed it and re-ran unigine-heaven. The test score improved from 28.75fps on 1.14 BIOS to 34.81fps on 1.20 BIOS. I can conclusively say that 1.20 bios update does not hinder performance.

Nice. I already decided to buy the E490. I'm only waiting the BlackFriday to see if prices gonna drop here in my country.

@mp035 can you explain to me what exactly _throttled_ does? With this BIOS update do I need to still use it? I'm gonna buy it without OS and install POP OS on it. Do I need any other software to achieve maximum performance? Fan speed control, maybe?

Dahdoul, 2019-11-21:
> First of all I can thank you @erpalma for that project. I just bought ThinkPad E590 with i7 and RX 550X and I must confirm that it was completely unusable together. Would you consider adding it to [...]

Hi @XeonBloomfield sorry for the late response, I followed the default configuration and everything is working well.

mp035, 2019-11-23:
@kafran The thinkpad has an embedded controller which restricts the CPU to around 65 degrees celsius.  On windows there is a daemon which periodically raises the CPU threshold to some software configurable limit.  This allows software to control CPU throttling behaviour.  If the software does not continuously update the CPU temperature threshold, the embedded controller returns it to around 65 degrees c.  Throttled simply takes the place of the windows daemon and periodically sets the temperature threshold to a reasonable level (default is around 90 degrees c).

kafran, 2019-11-24:
@mp035 Does the fingerprint reader works on Linux? Does the TPM 2.0 has any use on Linux?

mp035, 2019-11-25:
@kafran I don't have a fingerprint reader or TPM 2.0 so I can't help you there.

Dahdoul, 2019-11-25:
> @mp035 Does the fingerprint reader works on Linux? Does the TPM 2.0 has any use on Linux?

Unfortunately it doesn't work in Linux

kafran, 2019-12-19:
I finally got the courage and my E490 arrived. Boy it runs hot! @mp035 I was about to update the BIOS from 1.18 to 1.20 and it seems Lenovo has release a new BIOS update 1.22 which takes care of Thermal issue. I'm afraid to install this one (1.22) because of what happened on E480 with BIOS 1.14 (https://www.notebookcheck.net/ThinkPad-E480-i5-8250U-RX-550-Laptop-Review.280524.0.html). Has anyone tried the new BIOS 1.22? It seems it raises the thermal limit to 80ºC.

360degreelove, 2020-07-29:
I am planning to buy E490 i7 8th Gen with 550 RX, but really confuses from this forum. Latest Bios shows V1.27 on Lenovos website (https://support.lenovo.com/ae/en/downloads/ds506071). Is the throttling and heating solved?

kendrich123, 2024-11-28:
Can someone explain or teach me how to solve this issue, I didn't get it by reading the comments, badly needed please @mp035  @erpalma