erpalma/throttled #122: How did you get T480 run prime95 at 44W under Windows? I can only reach 15W

clouddemo1, 2019-06-18:
I have been reading your reddit post: https://www.reddit.com/r/thinkpad/comments/870u0a/t480s_linux_throttling_bug/

You mentioned you can get the T480 run prime95 at 44W under Windows a year ago.

But now it appears my T480 is throttled at 15W. It reaches 22W for a short moment then XTU shows Power Limit Throttling then it drops to 15W. 

My T480's PSU is 65W.
CPU is i7-8650u.
GPU is MX150.
I'm using Windows 10 with the newest BIOS 1.24.
It appears some people have similar issues: https://www.reddit.com/r/thinkpad/comments/935isi/t480s_power_limit_throttling_after_clean_install/
They said unstall the Lenovo PM Driver or Disabling Lenovo Intelligent Thermal Solutions can solve the issue.
Other people https://www.reddit.com/r/Dell/comments/a27e1l/finally_permanently_fixed_dell_inspiron_power/ suggested unstalling "Intel Dynamic Platform and Thermal Framework" driver can solve this.

I did all of these. I unstalled all Intel DPTF staffs all Lenovo staffs and disabled the DPTF service and all Lenovo services. But it does not work.

After deleting all these, the XTU always shows the Thermal Throttling. Then the CPU won't go up beyond 70 C。。

So how to remove all throttle limits to get max power of the T480 at Windows?

clouddemo1, 2019-06-18:
I have also tried set power plan to "balanced" or "ultimate performance". But it appears no difference.
My best result for now is 
1. Uninstall all intel DPTF stuffs. But not Lenovo stuffs.
2. Reboot.
3. Set Lenovo Intelligent Thermal Solutions service to Manual and stops it. But everytime after reboot, it have to start and stop this service manually for once...

This way I can get it running at 25W, CPU temp 93 C, and XTU only shows Power Limit Throttling.

erpalma, 2019-06-18:
Yes, DPTF started limiting TDP on Windows too after some time. Right now when I need to boot into Windows I just fire ThrottleStop, without XTU. This way I can get 44W stable.

clouddemo1, 2019-06-19:
> Yes, DPTF started limiting TDP on Windows too after some time. Right now when I need to boot into Windows I just fire ThrottleStop, without XTU. This way I can get 44W stable.

Thank you. I just tried ThrottleStop, it shows PROCHOT is 86 C。Where to increase this temp?  Is this temp is set by Intel DPTF or Lenovo stuffs? Did you uninstall any drivers?

clouddemo1, 2019-06-19:
After digging a bit deeper. I found that starting and stop the Lenovo Intelligent Thermal Solutions service manually actually set the MSR_TEMPERATURE_TARGET=0x1A2 bit from 0xE164 to 0x0364. Meaning the thermal Throttling is changed from 70C to 97C as explained here: https://forum.51nb.com/forum.php?mod=viewthread&tid=1889105

This works on the this T480. But not working on T470...
I also tried using RWeverything, but I get a blue screen error caused by RwDrv.sys...

EternalPhane, 2019-11-23:
@clouddemo1 I even tried writing a Windows kernel driver myself. Everything's fine with `__readmsr`. However at the very moment I touch `__writemsr`, a GSoD always occurs.

EternalPhane, 2019-12-09:
I think this can be closed with Windows 10 20H1, which changed the behavior of Virtualization-Based Security, undocumented however. [See this post](https://forums.lenovo.com/t5/ThinkPad-X-Series-Laptops/Lenovo-X1E-still-throttling-at-80c-when-Hyper-V-enabled/m-p/4594082/highlight/true#M108215).

bybor, 2023-05-26:
> I think this can be closed with Windows 10 20H1, which changed the behavior of Virtualization-Based Security, undocumented however. [See this post](https://forums.lenovo.com/t5/ThinkPad-X-Series-Laptops/Lenovo-X1E-still-throttling-at-80c-when-Hyper-V-enabled/m-p/4594082/highlight/true#M108215).

The post was deleted and internet archive doesn't have it.

AFAIK, i5 (8250u) cannot consume 44W, only i7 can. i5 can do around 30W, which can be cooled by T480 cooling system (with 2 heat pipes, not sure about one with a single pipe) and by T480s's. 15W is a limit set by something and can be changed under windows 10 with throttlestop. 

It's offtopic maybe, came here from search, but i hope it may give some context to discussion

rehanzo, 2023-09-06:
> > I think this can be closed with Windows 10 20H1, which changed the behavior of Virtualization-Based Security, undocumented however. [See this post](https://forums.lenovo.com/t5/ThinkPad-X-Series-Laptops/Lenovo-X1E-still-throttling-at-80c-when-Hyper-V-enabled/m-p/4594082/highlight/true#M108215).
> 
> The post was deleted and internet archive doesn't have it.
> 
> AFAIK, i5 (8250u) cannot consume 44W, only i7 can. i5 can do around 30W, which can be cooled by T480 cooling system (with 2 heat pipes, not sure about one with a single pipe) and by T480s's. 15W is a limit set by something and can be changed under windows 10 with throttlestop.
> 
> It's offtopic maybe, came here from search, but i hope it may provide additional context to this issue

Thanks for that info, so would it make sense to just set it to 30W instead of the 44W?

bybor, 2023-09-07:
I don't know. My understanding was that `throttled` (this tool) cannot set value of your choice, but rather uses some constant. Doing so, it fixes the most annoying issue - removes 15W limit imposed by Lenovo, and it's a game changer.

Under Windows I've been using another tool - ThrottleStop which provides a GUI for setting TDP values (and a lot more, with a huge potential to waste your time ;) ). I spent some time picking the values so that my laptop doesn't go into thermal throttling while running at max possible speed. Here's how I got into this 30W vs 44W thing. The point of limiting the max TDP is that you can have a horizontal line of constant 26W consumption as long as you can (under high load), but if you let the CPU do this, it will just go "spikes" from 30 to 20 and back (I took these values out of my head) with average power consumption and performance being the same. The feeling of controlling the situation is pleasant, but that's it, IMHO. 

So, I would rather stick with max performance allowed and let CPU's internal logic do its work.